goecharger/go-eCharger-API-v1 #3: Undocumented json fields

ttudor, 2020-08-18:
Firmware 0033. In the status API I have the following undocumented fields:

```
{ 
...
"tma": [57.50,48.13,52.0,78.12]
"amt": "23"
...
}
```

It would be great to have these added in a newer revision of the API doc

hofster, 2020-08-18:
„tma“: internal temperature sensor 0-3
„amt“: max ampere limited through temperature sensors (32 = no limit)

ttudor, 2020-08-18:
Great, thanks ! It explains the problems I see with my charger. Would you add them to the doc or would you prefer to close and just leave the info here ? Please feel free to close this issue or tell me to.

hofster, 2020-08-18:
I’m not affiliated with go-e. I asked the go-e support about these values some time ago and just wanted to share the information ;-)

Regarding the temperature values:  At the moment (outside temperature is 20.0 C, no charging) my go-eCharger reports these values:
TME1: 35.17 C
TME2: 34.42 C
TME3: 34.59 C
TME4: 34.87 C

Yours seem to be a little high.

ttudor, 2020-08-18:
Yes, this happens while charging, which causes current limitation. I will keep this comment open and note here if I get relevant info from go-e.

At this very moment, outside temp 21.0C, and in stand-by (car not connected) I get:

TME1:  23.00 C
TME2: 22.37 C
TME3: 22.62 C
TME4: 22.25 C

Writing this down just in case somebody else stumbles upon this.

chk2902, 2020-10-14:
Cam "AMT" be set? On my device, it's 32 A, which is senseless as it's the limit anyway.

peterpoetzi, 2020-10-14:
amt cannot be set, but it will be lower if the go-eCharger is overheated