goecharger/go-eCharger-API-v1 #5: Changing frequency of publishing MQTT messages

rhuss, 2020-08-28:
Is it possible via the API (or other means) to set the frequency how often MQTT message are published ? 5s seems to be a bit fast for me as I'm storing every value in a database, but there are not many changes within 5s. E.g for me 30s or 1 minute would be totally fine, too.

MK-2001, 2020-08-28:
Hi,

There is an adapter for IOBroker. There you can adjust the frequency in the settings. And influx is available on ioBroker as well.

https://github.com/MK-2001/ioBroker.go-e

rhuss, 2020-08-29:
Thanks for the heads up !

I'm not using ioBroker but use MQTT directly in my home automation, in this case the Telegraf to import mqtt messages to influxdb. However the frequency how often mqtt status messages are send need to be configured on the publisher's side. 

Is IOBroker also using mqtt as protocol for obtaining the status ?

MK-2001, 2020-08-29:
Hi, ioBroker is using the HTTP Interface. But it could be done with a feature request there.

rhuss, 2020-10-11:
@peterpoetzi Is there anything planned for a configuration option to change the frequency when sending out MQTT messages ?

rhuss, 2020-10-11:
@MK-2001 My problem with the HTTP interface that is not encrypted, and even worse, not even secured with user / password. Not something that I really want to use, even when connected only to my local WLAN network.

rhuss, 2020-10-11:
Also, I wonder whether there is a more active community around the programmatic usage of go-eCharger. I really love that it is that open for management, e.g. currently I'm planning an adaptive load management based on the data that I receive from my PV device, so depending on how much current my PV generates I want to dynamically adjust the charging power so that it uses as much as possible energy from my own PV when charging my car.

peterpoetzi, 2020-10-15:
Hi, it is planned but currently not implemented, so I will leave this issue open

rhuss, 2020-10-15:
@peterpoetzi thanks for coming back on this. If there is something I can help to test or otherwise help, please let me know.

utisch, 2020-12-29:
@peterpoetzi :  Yes, would be great to get this configuration option.
With my fhem home automation the 5sec is reducing the lifetime of my flash-disk a lot - as I cannot suppress all the logs which are written to the flash.

rhuss, 2022-03-03:
@peterpoetzi any news on if and possibly when we could have this feature? It would be also helpful if the default period would be substantially longer than 5s.

rhuss, 2023-02-15:
@peterpoetzi any update (even a negative one) would be appreciated. Or is there any other way to reduce the frequency of mqtt messages ?

utisch, 2023-10-24:
I would also appreciated an answer from "go-e".  Waiting for this since more than 3 years.
The high update frequency of 5sec is really reducing the functionality of the go-e a lot

utisch, 2023-10-25:
Hello 0xFEEDCODE64,
   Yes, that's exactly what I'm looking for:  a configurable MQTT update period (currently set to 5sec).

   I have a go-e charger V4.

  Do you already know when this next release will be out ?

Thanks a lot in advance ...

rhuss, 2023-10-29:
> Just to confirm, you are asking for a configurable mqtt data intervall for the first generation of go-e chargers, not the "V3" or "V4" / gemini chargers?

Exactly, I still own a go-e charger from the first generation.

drgrandios, 2024-04-07:
> Just to confirm, you are asking for a configurable mqtt data intervall for the first generation of go-e chargers, not the "V3" or "V4" / gemini chargers? for newer models starting with V3 and V4, we actually have this already planned in and will be part of the next release

Any news on this? My Gemini charger sends > 10 messages every second and I would be happy if this could be reduced...

hessijames79, 2024-04-26:
I would also appreciate a fix.

utisch, 2024-04-27:
Same here: 
  A configurable MQTT cycle time would be great.
  I have a go-e V4