goecharger/go-eCharger-API-v1 #6: Correction necessary?

Clemens-XXX, 2020-09-16:
Die API ist nicht vollständig dokumentiert - folgende Punkte fehlen bzw. sind falsch:

tmp - immer "0", wurde durch das temperatur-array tma ersetzt

tma - array aller 4 Temperatursensoren in °C (z.B. [24.00,24.12,23.38,23.38])

amt - die maximale Amperezahl wenn diese durch die Temperatursensoren limitiert ist (bzw. "32" wenn es nicht limitiert ist)

txi - das transmit interface und in der Home+ immer "0"

Source: https://www.loxwiki.eu/pages/viewpage.action?pageId=72122962

hofster, 2020-09-16:
Das ist nicht richtig:

tmp zeigt bei mir einen Wert ungleich 0 an und zwar in der Nähe der 4 tma-Werte. Auch txi ist bei meinem HOME+ nicht 0, sondern 2.

(Firmware 033)

SebastianDietrich, 2020-10-19:
Bei mir ist tmp und txi immer 0 (Firmware 033)
Lt. App ist keine neue Firmware Version verfügbar - das stimmt aber auch nicht (aktuelle Firmware 040), die hat auch einen neuen Befehl amx (setzt Ladestrom ohne ihn ins EEPROM zu speichern)

peterpoetzi, 2020-10-19:
040 ist jetzt für alle freigegeben