goecharger/go-eCharger-API-v1 #7: MQTT verbessern?

Clemens-XXX, 2020-10-05:
Schaue mir gerade Optionen für den lokalen MQTT an (geniales Feature!)

Ich sehe keinen Sinn darin, bei MQTT parallel zu Übermittlung der Einzelwerte noch einmal alle Werte als Array zu senden. Konkret: go-eCharger/XXXX/status. Zumindest hier zerhaut mir das die Anzeigen. Vielleicht seht Ihr das auch so und könnt das abschalten?

peterpoetzi, 2020-10-05:
Der go-eCharger sendet über MQTT eigentlich jeden Wert nur einmal, eventuell ist das ein Anzeigeproblem beim MQTT Client? Wie sieht denn die Anzeige aus?

Clemens-XXX, 2020-10-05:
[image]
So stellt sich das bei mir im MQTT Broker dar. Erst ein Array ähnlich wie bei JSON und dann alle Werte einzeln. (Loxberry mit MQTT Plugin zu Anbindung an Loxone Hausautomatisierung).

peterpoetzi, 2020-10-05:
Diese einzelnen Topics kommen nicht vom go-eCharger, sondern werden wohl von MQTT Broker angelegt um das JSON Objekt aufzusplitten

Clemens-XXX, 2020-10-06:
Hm, also andere MQTT-Quellen machen das nicht. Und ich habe einige eingebunden ....

closed 2020-10-08 by peterpoetzi