goecharger/go-eCharger-API-v1 #11: Documentation is missing details or confusing about REST API

andrei-niciu, 2020-11-15:
1. What HTTP method is needed to change a parameter ? POST, PUT or PATCH ? At page 13 it is mentioned that Method is SET. However, this is not a HTTP method so I assume it just refers to what is possible to do with parameters (in this case, setting a new value).
2. From the example, it doesn't seem like there is a way to change multiple parameters at once in the same HTTP request. Is there by chance a way ?
3. Regarding rate limiting: would the device answer with 429 "Too many requests" ?

dumdiedum, 2020-11-22:
Regarding points 1 and 4: I've tried to change the lbr value (led brightness): 
http://x.x.x.x/mqtt?payload=lbr=100 is the right url. The funny part is: I tried several methods like PUT, SET (just used this as method value although I know there is no http method like SET) and also GET 😂 It worked every time!

peterpoetzi, 2020-11-27:
1. All HTTP Methods are GET. This is to allow users to try the requests with their web browser, and there are also some industrial PLC and other systems, which can only send GET requests. 

2. It is currently not possible to change multiple parameters at once.

3. Rate limiting is currently disabled and will be removed soon.

4. This is is not possible at the moment. You need to send 2 HTTP requests

closed 2020-11-27 by peterpoetzi