goecharger/go-eCharger-API-v1 #15: Where is the value of tmp measured?

klacol, 2020-12-06:
I have integrated our go-eCharger with this great [integration](https://github.com/cathiele/homeassistant-goecharger) into [Home-Assistant](https://www.home-assistant.io). 

The payload of the API deliveres the value of the field `tmp`. This is defined in the documentation as:

> Temperature of the controller in °C

Today the value shows 4-6 °C. I am wondering where this tempature is measured, since my go-eCharger is mounted in my cellar room (in the heating room). The room is not so cold. The cable is plugged in and the main part of the cable is placed outside. Is it possible, that the temperature is measured somewhere in the cable?

klacol, 2020-12-06:
The value of the temperature is measured between 3 to 6 °C, which seem to bee reasonable for outside temperature today (Western Germany in December).

[image]

peterpoetzi, 2020-12-06:
This is the data from the temperatur sensor on the microcontroller's silicon die, which is not well calibrated and can be off by multiple counts. Maybe, you also have the tma field in the status object, which contains more accurate temperature readings (only available for newer charging stations)

klacol, 2020-12-06:
Ok, thanks for the clarification. No, the charging station is from the beginning of 2019 and has no `tma `field.

closed 2020-12-06 by klacol

klacol, 2021-03-28:
see also the issue #6.