goecharger/go-eCharger-API-v1 #16: JSON-Value "tmp" für Temperatur zeigt komische wechselnde Werte

rleins, 2020-12-07:
Hallo,

ich würde gerne die Temperatur anzeigen lassen. Dafür gibt es das Feld "tmp", welches die Temperatur (ich glaube des Chips oder des Boards) angeben soll.
Dieses zeigt bei mir meistens 0, wechselt aber auch gerne mal auf -15 oder 10.

Auf jeden Fall ist das nicht die Temperatur, die wirklich herrscht.

Liegt bei mir ein Hardware-Defekt vor, oder ist dieser Wert nur nicht korrekt ausgelesen und im JSON-String damit fehlerhaft?

Grüße
Reiner

peterpoetzi, 2020-12-07:
Hi, der Temperatursensor ist nicht kalibriert und sehr ungenau. Neuere Ladeboxen haben bessere Temperatursensoren die über den tma Parameter als Array ausgelesen werden können, die sind viel genauer

rleins, 2020-12-07:
Hi,

ok, danke, die Werte sehe ich (vier Werte).
Welcher Wert steht denn für was? Das fehlt noch in der Schnittstellenbeschreibung.

Grüße
Reiner

KaiDK, 2021-01-17:
Schau mal hier: https://github.com/goecharger/go-eCharger-API-v1/issues/6#issue-702531026

> tma - array aller 4 Temperatursensoren in °C (z.B. [24.00,24.12,23.38,23.38])

man10to, 2021-02-10:
Hallo,
Ich habe drei Charger alle mit Firmware 0.40
Bei Tmp zeigen alle nur einen Wert an der immer zwischen der ungefähren Temperatur und meist 0 wechselt.
Hab ich dann alte Hardware (geliefert im Dezember 2020) oder kann man das auch auf die 4 Werte bekommen ?