goecharger/go-eCharger-API-v1 #17: Modbus Register doppelt

edgale, 2020-12-08:
Hallo,
das Modbus Register 315 ist in der Doku 2 mal vergeben.
Rückt alles einfach eine Adresse nach hinten?

https://github.com/goecharger/go-eCharger-API-v1/blame/master/go-eCharger%20Modbus%20TCP%20API%20v1%20DE.md#L79

edgale, 2020-12-08:
Und Register ```126``` ist als ```Input Register``` in ```4. Registertypen```  angegeben, taucht aber nicht als Regsiter in ```3. Modbus Register``` auf. Auserdem fehlt Regster ```120``` in ```4. Registertypen```

rootandy, 2021-01-03:
Hallo,

so die meisten Punkte sind jetzt soweit alle behoben.

edgale, 2021-01-04:
Hallo,

danke für die Änderungen.
Das Register 315 is allerdings in der Doku immer noch doppelt belegt.
[image]

closed 2021-09-17 by peterpoetzi