goecharger/go-eCharger-API-v1 #20: Neuer Parameter "amx" nicht im Status oder Set-Antwort

hombach, 2020-12-11:
> 040 ist jetzt für alle freigegeben

Leider wird "amx" nicht im Status zurückgegeben.... der Wert von amx kommt bei mir in der Antwort aber als "amp" zurück. Ist bei mir allerdings kurz darauf wieder auf Stand des alten "amp" - denke aber das liegt an meinem code (?)

peterpoetzi, 2020-12-11:
Hi, ja amx ist kein "Wert" der auf der Ladebox gespeichert ist, die Ladebox kennt nur amp. Wenn man amx einmal setzt, kommt man an dem in Flash gespeicherten Wert nur wenn man die Ladebox neu startet

hombach, 2020-12-11:
Wäre aber schon hilfreich wenn amx nach dem setzen in der Antwort enthalten wäre.... ;)
Kann bei mir auch noch keinen Fehler finden warum die Wirkung von amx nur ein paar sekunden hält, bin aber noch auf der Suche.

hombach, 2020-12-11:
Issue kann IMHO geschlossen werden - in der Set-Antwort würde ich "amx" allerdings schon gerne sehen :)

closed 2020-12-11 by peterpoetzi