goecharger/go-eCharger-API-v1 #24: How to decrypt api response

tillschweneker, 2020-12-23:
Hi,

when I use E2E-encryption and try to access the status via the cloud api, I get an encrypted response.

How can I decrypt it? Does anyone have an example for doing this with the crypto package of nodejs?

When I try the following decrypt method, I always get an error "ERR_CRYPTO_INVALID_IV".
`hash.iv` contains `status.iv` from the response and `hash.content` contains `status.aes`.
```js
const algorithm = 'aes256';
const secretKey = '*secret-passphrase*';

const decrypt = (hash) => {
    const decipher = crypto.createDecipheriv(algorithm, secretKey, Buffer.from(hash.iv, 'hex'));

    const decrypted = Buffer.concat([decipher.update(Buffer.from(hash.content, 'hex')), decipher.final()]);

    return decrypted.toString();
};
```

peterpoetzi, 2020-12-23:
Hi,

This function is used in the go-e app:

```
function decrypt(cipher, iv, mac, version, sse) {
    var esp8266_msg = cipher;
    var esp8266_iv = iv;
    var keys = getActualKeys(sse);
    if (keys.key == null || keys.hmac == null) {
        return {
            "version": "A"
        };
    }
    var plain_iv = new Buffer(esp8266_iv,'base64').toString('hex');
    var iv = CryptoJS.enc.Hex.parse(plain_iv);
    var plain_key = new Buffer(keys.key,'base64').toString('hex');
    var key = CryptoJS.enc.Hex.parse(plain_key);
    var plain_hmac_key = new Buffer(keys.hmac,'base64').toString('hex');
    var hmac_key = CryptoJS.enc.Hex.parse(plain_hmac_key);
    var plain_cipher = new Buffer(cipher,'base64').toString('hex');
    var cipher_arr = CryptoJS.enc.Hex.parse(plain_cipher);
    var iv_cipher = [];
    iv_cipher.push.apply(iv_cipher, wordArrayToByteArray(iv));
    iv_cipher.push.apply(iv_cipher, wordArrayToByteArray(cipher_arr));
    var hmac = CryptoJS.HmacSHA256(byteArrayToWordArray(iv_cipher), hmac_key);
    var hmac_b64 = CryptoJS.enc.Base64.stringify(hmac);
    var hmac_correct = (hmac_b64 == mac);
    var bytes = CryptoJS.AES.decrypt({
        ciphertext: CryptoJS.enc.Base64.parse(esp8266_msg)
    }, key, {
        iv: iv,
        mode: CryptoJS.mode.CBC,
        padding: CryptoJS.pad.Pkcs7
    });
    var plaintext = bytes.toString(CryptoJS.enc.Base64);
    var decoded_msg = new Buffer(plaintext,'base64').toString('ascii');
    var ret = {
        "version": "A"
    };
    if (hmac_correct) {
        try {
            decoded_msg = decoded_msg.replace(/[^\x20-\x7F]/g, "");
            ret = JSON.parse((decoded_msg));
        } catch (e) {
            console.log("exception " + e);
        }
    } else
        console.log("HMAC incorrect");
    return ret;
}
```

Here are the keys generated (from the variable this.pbkdf_key):

```
var hash = CryptoJS.PBKDF2(this.pbkdf_key, "c378cbd27caa47f38c3395e6f32656a4_"+last_config.sse,
                            { keySize: 512/32, iterations: 3000 });
var ba=wordArrayToByteArray(hash.words,32);//not all 80 are needed
var k1=ba.slice(0,16); //128bit key for aes
var k2=ba.slice(16,32);//128bit key for HmacSHA256
var basek1= CryptoJS.enc.Base64.stringify(byteArrayToWordArray(k1));
var basek2= CryptoJS.enc.Base64.stringify(byteArrayToWordArray(k2));

saveKeys(last_config.sse,basek1,basek2);
```

It requires crypto.js

Best regards,
Peter Pötzi

tillschweneker, 2020-12-23:
Hi Peter,
thanks for the quick response!

But isn't there some information missing, though? 

The response from the API only has the relevant keys `version`, `iv`, `mac` and `aes`.

In your decrypt function the `cipher` property refers to `aes`, right? 
But what is `sse`?

And in the function there is a call to `getActualKeys`. Where does this function come from?

Thanks,
Till

peterpoetzi, 2020-12-23:
Hi,

for the function `decrypt(cipher, iv, mac, version, sse)`, fill it with `(aes, iv, mac, version, sse)` from the json response. sse is the Serial number of the charger, i.e. 000001

The `getActualKeys `method has a database in the background, that would be filled with `saveKeys`. You can omit the database and directly use `(basek1,basek2)` as `keys={key:basek1,hmac:basek2}`

Best regards,
Peter Pötzi

tillschweneker, 2020-12-23:
Hey,

alright, but what are `basek1` and `basek2`? 

BR,
Till

tillschweneker, 2020-12-23:
Oh, and can you also provide the functions `byteArrayToWordArray` and `wordArrayToByteArray`?

peterpoetzi, 2020-12-24:
Sure
```
function wordArrayToByteArray(wordArray, length) {
	if (wordArray.hasOwnProperty("sigBytes") && wordArray.hasOwnProperty("words")) {
		length = wordArray.sigBytes;
		wordArray = wordArray.words;
	}

	var result = [],
		bytes
		i = 0;
	while (length > 0) {
		bytes = wordToByteArray(wordArray[i], Math.min(4, length));
		length -= bytes.length;
		result.push(bytes);
		i++;
	}
	return [].concat.apply([], result);
}
function byteArrayToWordArray(ba) {
	var wa = [],
		i;
	for (i = 0; i < ba.length; i++) {
		wa[(i / 4) | 0] |= ba[i] << (24 - 8 * i);
	}

	return CryptoJS.lib.WordArray.create(wa, ba.length);
}
```

tillschweneker, 2020-12-25:
Perfect thanks @peterpoetzi! I think I can get it to work now. 

Happy holidays!

closed 2020-12-26 by peterpoetzi