goecharger/go-eCharger-API-v1 #27: Wie kann ich die Ladung über API komplett stoppen

AlxBe, 2020-12-31:
Hallo @peterpoetzi,
danke für die Dokumentation! 
Soweit ich verstanden habe kann ich meinen PV geführte Ladung über API und  setzen der "amx" Werte regeln.

Wie ist es aber möglich, dass ich die Ladung komplett stoppe (z.B. nachts oder während eines gewitters)? Dazu habe ich keine wirkliche Lösung gefunden. Wesentlich wäre, dass ich den go-e nicht so blockiere, dass ich irgendwas an einer SW machen muss.

Mir schwebt folgendes vor: Wenn ich den go-e auf einen Wert < 10 A stelle und das Auto anschließe, erkennt das die SW und steuert PV geführt (Überschussladen). Wenn kein Überschuss auf der PV mehr ist, schalte ich den Ladevorgang auf 0 (keine Ladung). Habe ich wieder einen Überschuss geht es weiter (z.B. nächster Tag). Will ich fahren klemme ich das Auto ab und go-e ist wieder im Normalzustand (insbesondere muss ich nicht manuell Aktivieren).

Kann ich bedenkenlos (insb. ohne den Flash auf Dauer zu ruinieren) die Farbe der LEDs ändern, wenn ich im PV geführten modus lade?

Danke für die Antwort.

Grüße

fuchs-1978, 2020-12-31:
Dazu gibt es alw, damit kannst du die Ladung unterbrechen

Homoran, 2021-02-21:
Wird eine Änderung des alw ins flash geschrieben?
Eine Nutzung um reine PV Überschussladung durchzuführen würde bei mehrmaligem ab- und wieder anschalten das Budget von 100.000  Schreibvorgängen arg belasten

fuchs-1978, 2021-02-21:
Wie oft schreibst du das denn an einem Tag?

Homoran, 2021-02-21:
Hab die Box erst seit paar Tagen.
Heute lief meine erste gesteuerte Ladung. Zum Glück war schönes Wertter UND im Haus waren leine größeren Verbraucher aktiv.
[image]

Wenn jedoch Wolken oder Verbraucher den Überschuss runterziehen, kann das schon ein paarmal am Tag sein.

Aber schreibt er das jetzt in den Flashspeicher oder nicht?

fuchs-1978, 2021-02-21:
Das kann ich dir leider nicht beantworten, aber teste es doch mit alw=1 und dann machst du einen Neustart. Dann nochmal mit 0. Wenn der Status jeweils bleibt war es im Eeprom. 
Ich lasse bei mir sie Ladung mit 6A weiterlaufen. Ich würde mir dann eher Gedanken um die schütze im charger und im Auto machen wenn ich das oftam Tag schalte.

Homoran, 2021-02-21:
Da wäre es doch für alle besser, wenn man es mit amx=0 steuern könnte.
Leider geht das nicht

Wildbill-Z, 2021-03-02:
amx=0 ohne Schreiben in den Flash um das Laden abzubrechen fände ich in der Tat auch besser, als mit alw=0 jeweils einen Schreibzugriff in den Flash zu machen. Das Setzen von alw kommt bei mir zwar maximal zweimal am Tag vor (einmal an, einmal aus), so dass ich das theoretisch sehr lange und oft machen könnte, bis die Schreibzyklen aufgebraucht wären, aber wenn es sich machen liesse, diese so ganz zu unterbinden, warum nicht? Standardwert nach Strom weg könnte ja sein, dass man Laden will mit dem Wert der in amp gespeichert ist.

Gruss, Jürgen

chk2902, 2021-06-04:
Bei PV-Ladung kann das viele Male am Tag vorkommen.

Die Schütze haben eine Haltbarkeit von etwa 50.000 Schaltvorgängen.

Ich lade schon seit 2 1/2 Jahren per PV und habe ca 3.000 oder mehr Ladevorgänge, ohne Probleme deswegen.