goecharger/go-eCharger-API-v1 #35: Writing MQTT value locally

torebrigaglia, 2021-02-03:
Hi Thank you for this charger, it was missing in the market.
it seems i can only set values on the charger with modbus or http://x.x.x.x/mqtt?payload=
i am trying to publish a value to be set using this topic:  "go-eCharger/serialnumber/cmd" or "go-eCharger/serialnumber/cmd/req" publishing a json formatted message like {"amp":"10"}  but it has no effect. 
Please advice

erikderzweite, 2021-04-25:
Neither does escaping work, nor "amp=10" or similar. An example in the documentation would be nice.

erikderzweite, 2021-04-25:
You need to provide the command in the same format as with modbus or REST api, e.g. "amx=10"

erxbout, 2024-07-03:
Exactly.. In the API V1 this is not handled by json input

closed 2024-07-03 by erxbout