goecharger/go-eCharger-API-v1 #37: Wert von AMX nicht in den MQTT Daten

Dreistein19, 2021-02-19:
Vergessen ? oder hab ich das übersehen.
Version 0.40 ist Installiert.

Therealsipple, 2021-02-25:
Nein, nicht übersehen, er ist nicht vorhanden.
Im Moment ist mangels Rückmeldung vom Support noch nicht einmal klar, ob AMX tatsächlich in der 040 existiert. 
Wenn das der Fall ist, dann wohl nur als "unsichtbarer" Schreib-Parameter. 
AMX wäre einer der wichtigsten Parameter um sich bei PV-gesteuerter Ladeleistung nicht in kürzester Zeit die Speicherzelle kaputt zu schreiben. AMX sollte aber auch unbedingt im Status ausgegeben werden, um bestehende Regelsoftware ganz einfach von AMP auf AMX umstellen zu können.

fuchs-1978, 2021-02-25:
Amx kannst du aber bei der http API auch nicht lesen. Mal probiert ob schreiben geht? Man kann nur AMP auslesen, egal ob AMP oder AMX gesetzt wird.

Therealsipple, 2021-02-25:
Richtig.
Der Status JSON String ist eh bei beiden Varianten identisch.
Umgekehrt wäre es mir lieber. Den AMP gar nicht erst anzeigen und dafür den AMX. Nach Reboot übernimmt ja der AMX den Wert von AMP (laut Beschreibung). Macht Sinn. 
Bei Gelegenheit werde ich das wohl mal testen, wenn keine Antwort vom Support kommt.