goecharger/go-eCharger-API-v1 #43: Eindeutiger hostname?

tuxedo0801, 2021-03-18:
Hallo,

go-eCharger melden sich beim DHCP Server mit "go-echarger" als Hostname an. Der Hostname wird vom Router in den DNS eingetragen (machen vor allem viele Konsumer-Router so...), so dass ich über den Hostnamen die Wallbox künftig erreiche. 
Problem: Ich habe zwei Geräte in der Garage hängen. Und beide fragen mit dem identischen Hostnamen beim DHCP eine IP an. Und der identische Name kollodiert dann in den DNS Einträgen. Mit der folge: Nur eins von den beiden Geräten erreiche ich per Hostname (der der zuletzt kam). 
Ich müsste von Hand eigene Namen im DNS vergeben. Unschön, wo es doch dieses tolle Feature über den DHCP gibt...

Zwei Optionen wären vorstellbar:

1) API-Kommando zum anpassen den Hostnamens: Dann kann man seinen gewünschten Namen selbst am Gerät eintragen.
2) Default-Hostname um Seriennummer erweitern. z.b. "go-echarger-012345", dann wäre der Name wieder eindeutig.

Freue mich über Feedback.

LichtiMC, 2021-03-27:
Lösung: DHCP-Reservierung für die beiden Charger einrichten und IP verwenden (da kann man oft gleich den Hostnamen manuell vergeben) oder DNS einmalig anpassen wie von dir vorgeschlagen.

Die Erweiterung des Hostnamens steigert nur unnötigerweise die Komplexität desselben was grundsätzlich gegen das Konzept von DNS spricht. (Und bei den allermeisten wirds passen so wie es jetzt ist, da sie nur einen Charger haben.

tuxedo0801, 2021-03-27:
> Die Erweiterung des Hostnamens steigert nur unnötigerweise die Komplexität desselben was grundsätzlich gegen das Konzept von DNS spricht.

es spricht ja nix dagegen die Charger mit alle dem selben Hostnamen wie bisher auszuliefern. Aber ich kann jedem Charger ja einen Namen geben, bzw. ihn umbenennen.
In meinem "Netzwerkpark" mit >50 Geräten hab ich - bis auf die go-e Charger - kein Gerät dem ich nicht in dessen Konfiguration einen Namen geben kann, der dann als "client name" im DHCP request mit benutzt wird. 

Da geht eigentlich gar nix gegen das Konzept von DNS und DHCP... DHCP sieht die Möglichkeit vor, dass sich ein Gerät mit einem Namen beim DHCP meldet (was der Charger ja heute schon macht). Und DNS sieht vor, dass ein Hostname in eine IP aufgelöst werden kann.

Und eben gang und gebe ist es, dass in lokalen Netzwerken in denen ein DHCP läuft, auch ein DNS vorhanden ist. Und damit niemand selbst an den DNS Hand anlegen muss, wird oftmals der Hostname aus dem DHCP Request zusammen mit dem IP-Lease in den DNS eingetragen. 

All das geht ja schon heute, auch mit dem Charger. Es würde in keinster Weise die Komplexität für diejenigen erhöhen die nur einen Charger haben, wenn man dem Gerät einen individuellen Hostnamen geben könnte. Und die, die einen halben Fuhrpark oder Ladepark haben, haben die Möglichkeit jedem Charger einen individuellen Namen zu geben. Der Vorschlag mit dem Anhängen der Seriennummer by default würde hier sicher helfen.

Und nochmal: Der 0815 User mit nur einem Charger in der Garage... dem ist der Hostname sicherlich völlig schnuppe. Aber all die, die mehr als einen haben wäre es sicherlich geholfen wenn man dem Charger, wie mit jedem anderen vernünftigen Netzwerkgerät, einen Namen geben könnte, den man für DHCP Request und Co. verwenden kann.

Für den Moment hab ich es mit manuellen statischen DNS Einträgen im DNS meines Routers gelöst. Ist aber die einzige Sonderlocke im Netzwerk mit >50 Geräten verschiedenster Hersteller und Geräteklassen.

LichtiMC, 2021-03-28:
Was du schreibst stimmt schon alles.
Ich gebe dir auch recht, dass es nett wäre, wenn der go-e charger Hostname anpassbar wäre, jedoch ist dies meiner Meinung nach ein niedrig zu priorisierender Feature-Request, da alles was dann möglich wäre auch jetzt schon möglich ist. 

Den Vorschlag den Hostnamen mit einer Seriennummer oder MAC (oder ähnlichem) zu erweitern würde ich jedoch entschieden ablehnen.
DNS wurde (unter anderem) erfunden um sich nicht sämtliche IPs aller Geräte merken zu müssen sondern jene mit einem einfach zu merkenden Namen ansprechbar zu machen. Wenn der Hostname nun also standardmäßig "goecharger-	8CAAB562A1F2" heißen würde, wäre der Sinn meines Erachtens damit dahin.

Jeder Ottonormal-Verbraucher freut sich, dass er seinen Charger ohne weitere Konfiguration mit go-eCharger ansprechen kann und Leuten mit 50 oder mehr Geräten im Netzwerk traue ich zu, dass sie ihr Netzwerk soweit im Griff haben, dass sie das "Problem" mit dem doppelten DNS-Eintrag lösen (lassen) können.

tuxedo0801, 2021-04-11:
Von mir aus aus kompatibilitätsgründen wie bisher so belassen, aber den Namen anpassbar machen. In der APP kann ich schon heute einen Namen vergeben. Aber das scheint nur in der APP zu passieren. Wenn ich hier den Hostnamen eingeben könnte würde das passen. Denn da sehe ich noch die ID des Chargers und kann dem Kind einen eindeutigen Namen geben. 

Zur Argumentation mit dem Hostnamen der sich an der ID orientiert "von Haus aus": Seeeehr viele Hersteller machen das in der Tat so. Viele WLAN Geräte kommen mit einem WLAN Namen ins Netz, der an ihre Mac-Adresse oder Seriennummer angelehnt ist, welche meist hinten aufgedruckt ist. Sehe das absolut nicht als Problem an. Otto-Normalverbraucher nimmt eh nur die APP und interessiert sich nicht für Hostnamen. Und wer einen ganzen Ladepark baut, wird sich freuen, wenn er irgend einen Bezug zur Seriennummer hat statt jeden einzeln einzuschalten, ihm einen ein-eindeutigen Hostnamen zu verpassen und dann den nächsten einzuschalten. 

Aber wie dem auch sei: Das mindeste wäre, der Namen des Geräts im Gerät zu speichern und dass dieses diesen Namen dann für DHCP Anfragen auch verwendet. 
Ein vorgegebener ein-eindeutiger Hostnamen würde aber Otto-Normalverbraucher auch nicht stören. Er sieht davon ja nichts und die App interessiert es nicht.

hackex, 2021-06-03:
> Aber wie dem auch sei: Das mindeste wäre, der Namen des Geräts im Gerät zu speichern und dass dieses diesen Namen dann für DHCP Anfragen auch verwendet.
> Ein vorgegebener ein-eindeutiger Hostnamen [...]

sehe das genauso. das wäre schon sehr hilfreich und ist bei vielen anderen Geräten schon Gang und Gebe.