goecharger/go-eCharger-API-v1 #44: go-e acquired by Fronius

geowiwi, 2021-03-29:
Now that go-e was seemingly acquired by Fronius, and they released their own version of the go-eCharger, will there be any furthere development of the firmware and the API documentation here? Or are the support and API going to be integrated into the Fronius ecosystem?

[Fronius Wattpilot](https://www.fronius.com/de/solarenergie/installateure-partner/technische-daten/alle-produkte/l%C3%B6sungen/fronius-wattpilot/fronius-wattpilot/wattpilot-home-11-j)

peterpoetzi, 2021-03-29:
It is not true that go-e was acquired by Fronius. The Fronius wattpilot is an OEM product manufactured by go-e and sold to Fronius International, as we also do for several other companies. The original go-eCharger will also soon receive new software with many new functions.

peterpoetzi, 2021-03-29:
There is no API documentation available for the Fronius Wattpilot and it is definitely not compatible with the go-eCharger API.

fuchs-1978, 2021-03-29:
can you publish something about the new features now?

peterpoetzi, 2021-03-29:
We are finishing a new app, and with the new app we will also have a completely new cloud infrastructure with the possibility to have historical data and APIs without rate limiting.

geowiwi, 2021-03-29:
Glad to hear and looking forward to new development from go-e @peterpoetzi.

closed 2021-03-30 by peterpoetzi

thecem, 2021-09-07:
Will be there an API for the Wattpilot? 
Is in FW 34.3 a API implemented but not activated?
Is there a Documentation for the Websocket?

Thanks a lot for saving a lot of re-engeneering...

ahochsteger, 2021-10-02:
@peterpoetzi maybe you're interested in my findings about how to communicate with the Fronius Wattpilot:
In short it uses JSON messages over websockets using the following URL: `ws://<local-WIFI-IP>/ws`.
More details can be found here: https://github.com/safepay/sensor.fronius/issues/48#issuecomment-932501077

techydude, 2024-05-19:
Hello Peter,

> Fronius wattpilot is an OEM product manufactured by go-e and sold to Fronius International, as we also do for several other companies

Can you please tell me which other companies re-brand the go-e EVSEs?
I'm wanting to control my Fronius Wattpilot with Home Assistant, but the OCPP integration doesn't appear to support go-e or Fronius (yet), so I wanted to check if any other currently supported EVSEs on this list (https://home-assistant-ocpp.readthedocs.io/en/latest/) are re-branded go-e?