goecharger/go-eCharger-API-v1 #45: AMX-Wert darf nur Gazzahlig (6..32) sein?

uklatt, 2021-03-29:
Hallo,

warum werden kein "Gleitkommazahlen" für AMX zugelassen?
Der PWM-Wert kann doch auch Zwischenwerte annehmen.
Das würde eine Regelung (z.B. nach PV-Überschuss) deutlich verbessern.

Momentan "verschenkt" man mindestens 0.5A*230V = 115W bei 1-phasigem Laden.
Bei 3-phasigem Laden sind es schon 345W.

Uwe

chk2902, 2021-06-04:
Interessant ist, dass er eigentlich prinzipiell nur alle 3 A ganzzahlig ist:

"Die SAE hat die maximale Stromlast auf der Basis einer Formel definiert, welche die 1000 µs Zykluslänge der Trägerfrequenz (das 1 kHz Signal) nimmt und je 10 µs Pulsweite mit 0,6 A multipliziert, um die Dauerlast des Anschlusses zu definieren (mit minimal 100 µs = 6 A und maximal 800 µs = 48 A)" (https://de.wikipedia.org/wiki/IEC_62196_Typ_2), oder auch in IEC "In der Norm IEC 61851-1 definiert der Bereich von 8 % bis kleiner 10 % Duty cycle die maximal verfügbare Leistung als 6 A, der Bereich von 10 % bis 85 % Duty cycle die maximal verfügbare Leistung als (% Duty cycle) multipliziert mit 0,6 A, der Bereich von größer als 85 % bis 96 % Duty cycle die maximal verfügbare Leistung als (% Duty cycle – 64) multipliziert mit 2,5 A, der Bereich größer als 96 % bis 97 % Duty cycle als 80 A"

uklatt, 2021-06-04:
Wo steht da etwas davon, dass der Strom nur ganzzahlige A-Werte annehmen darf?
Beim EVCC (nicht EVCC2) von ABLSursum kann/konnte man auch Zwischenwerte einstellen.
Für Überschussregelungen wäre eine geringere Auflösung sinnvoll.

Uwe

chk2902, 2021-06-04:
Nein, im Gegenteil: er ist nur alle 5 Stufen ganzzahlig:
6
6.6
7.2
7.8
8.4
9
...
zumindest gemäß o.a. Normen.

Aber stimmt: man kann das auch anders lesen als I = 6 + (Pulsbreite - 10) * 0.06, dann wäre das beliebig. I rest my case ;)

kaeferfreund, 2021-07-28:
@peterpoetzi ?

peterpoetzi, 2021-09-17:
Leider ist es bei der V2 nicht möglich Komma-Werte als Ampere zu verwenden. Eventuell wird es in Zukunft für die V3 als experimentelles Feature zur Verfügung stehen.

closed 2021-09-17 by peterpoetzi

uklatt, 2021-09-17:
Hallo Peter,

dummerweise habe ich eine V2 Wallbox...

Uwe

kaeferfreund, 2021-09-22:
Wichtiges Feature für uns mit insgesamt 7 Wallboxen im eigenen Hause!