goecharger/go-eCharger-API-v1 #48: Genauigkeit 'eto'

Chringbay, 2021-04-12:
Hallo, 

der Wert 'eto' ist beschrieben als: 
`energy_total:` Gesamt geladene Energiemenge in 0.1kWh - Beispiel: 130 bedeutet 13kWh geladen

Ich habe festgestellt, dass der Wert immer nur ganze kWh liefert, die letzte Stelle also immer eine '0' ist. 
Ist das so beabsichtigt? 

Hintergrund:
Ich betreibe eine Konfiguration mit OpenWb, diese errechnet die geladene Menge pro 5 Minuten anhand der Differenz der gesamt geladenen Energiemenge. Durch die niedrige Auflösung des Wertes habe ich bei der geladenen Menge große Ausschläge und die Grafik ist schwierig abzulesen.

Viele Grüße
Christian

chk2902, 2021-04-30:
Dasselbe übrigens bei den RFID-Zuordnungen.

marcelgsteiger, 2021-09-06:
Ich stelle dasselbe Problem fest. Ich berechne ebenfalls die geladene Menge pro 5 Minuten für eine PV-gerechte Abrechnung. Dazu verwende ich etwas wie:

(Math.floor(eto / 10) * 360000) + (dws % 360000 )

...das liefert mindestens bis vor Ende des Ladens brauchbare Werte. Nach Abschluss des Ladens kehrt das go-e die angebrochene Kilowattstunde einfach unter den Teppich und fängt dann mit dem auf 1 kWh "abgerundeten" eto und einem auf Null gesetzten dws wieder an. Deshalb kann man leider mit eto den Energiezähler nicht ersetzen, bei jeder Ladung bleibt im Mittel eine halbe Kilowattstunde ungezählt ;). Ich sehe deshalb vor, bei Statusänderung das dws getrennt für jeden Status zu speichern und dann bis 5 Minuten nach Ladebeendigung den zwischengespeicherten Wert aus dem Status "wird geladen" zu verwenden, damit der Negativsprung in der Gesamtleistung in die "ungenutzte Zeit" zwischen zwei Ladungen fällt und so in einem "nachgeschalteten Zähler" einfach ausgeblendet werden kann. So hoffe ich den Defekt ausbügeln zu können.

Grüsse
Marcel

basis42, 2021-10-20:
@chilobo Good point, ich denke dws könnte für meinen Case die Lösung sein, Danke!

marcelgsteiger, 2021-10-21:
@basis42 Vielleicht wurde auch abgerundet, weil ich inzischen die go-e neu gestartet hatte - weiss das nicht mehr so genau. Ich habe inzwischen eine weitere go-e in der Einstellhalle integriert, mit neuerer Hardwareversion und verwende damit API V2. In eto werden Wattstunden geliefert, und es scheint sich um einen brauchbaren Messwert zu handeln. Das macht alles viel einfacher.