goecharger/go-eCharger-API-v1 #50: cURL: Operation timed out

kaufi88, 2021-05-02:
Hi there,
we've three go-e charger for our students home.
To manage the wallboxes and charging activity we poll each wallbox every 5 minutes with cURL.

In some cases (about 2 of 5 times) we get an Operation timed out error.
URL:http://192.168.50.13/statusError:Operation timed out after 4062 milliseconds with 948 out of 1145 bytes received
URL:http://192.168.50.14/statusError:Operation timed out after 5000 milliseconds with 0 bytes received

Why does the wallbox stop transfering data?
How can this be fixed?

Thanks
Daniel

peterpoetzi, 2021-05-02:
I can confirm that the cloud service is overloaded at the moment. We are actively working on a new cloud service which will go live in the next weeks and will have a much higher capacity and no rate limiting / fair use policies.

kaufi88, 2021-05-02:
Thanks for the fast response.
As you can see from the IP, we use a the local API for polling the wallbox.
Does the slow cloud has an impact to the local API?

peterpoetzi, 2021-05-02:
Sorry, I did not look at that. The local API should not be affected by the cloud. Maybe the WiFi signal strength is too low. You can also try restarting the charger.

kaufi88, 2021-05-02:
Oh well, the WiFi of the box is really bad. I miss the RJ-45 jack.
Is there a way to check the wifi singal from the perspective of the wallbox?

(Yes, I already restarted the boxes...)

peterpoetzi, 2021-05-02:
I do not have any solution but a WiFi repeater at the moment. However it might be possible to make the connection better with a future firmware update, we are looking into that at the moment.

kaufi88, 2021-05-02:
Okay thanks. I'll try to get the accesspoint closer to the wallboxes and keep this topic updated when I've improved the WiFi signal.

closed 2021-09-17 by peterpoetzi