goecharger/go-eCharger-API-v1 #51: RFID beim Überschussladen

fuchs-1978, 2021-05-03:
Ich habe folgendes Problem. Zwei meiner Charger sind draußen, sind also so eingestellt, dass man sie mit RFID freischalten muss.

Nun habe ich das PV Überschussladen so realisiert, dass ALW die Box ein und ausschaltet. Dann würde sie aber auch freigeschaltet werden durch den Befehl wenn sie nicht vorher durch RFID freigeschaltet wurde.

Gibt es irgendwo ein Objekt welches ich dazu abfragen kann? Also RFID schaltet frei und wenn das Auto abgezogen wird ist die Box wieder gesperrt.

Habt ihr da Ideen?

fuchs-1978, 2021-05-03:
Man könnte es ja mit Unblocked by RFID-ID (uby) machen, oder gibt es noch etwas besseres?

l33tn00b, 2021-05-26:
Ich hijacke das hier mal mit einem ähnlichen Anliegen, das aufs Gleiche hinausläuft: Die API Spec liefert leider keine Info, wie/wann sich uby ändert.. Wenn ich also die Karte vor den Leser halte ändert sich UBY. Wird der Wert nun zurückgesetzt, wenn der Ladevorgang beendet ist? Wird der Wert zurückgesetzt, wenn das Laden abgebrochen wird? Hängt ein Zurücksetzen von der ALW ab? Ab wo zählt UBY? Null? Eins?

scaleupchris, 2022-03-09:
Ich wäre auch sehr daran interessiert, die RFID Karte per API setzen zu können um das Überschussladen auf die richtigen Zähler zählen zu lassen.
Dazu müsste man den Wert für uby irgendwie setzen können.