goecharger/go-eCharger-API-v1 #54: New Commands/Registers for Hardware V3 documentation needed

chk2902, 2021-05-31:
How to find out it's Hardware V3?
How to switch number of phases?
How to wake up BEVs like corsa, Peugeot or Zoe by the new method to switch off the charge signal? Or is that automatically done by the new firmware?

hackex, 2021-06-03:
I would also like to know the new API commands for hardware version 3.

chk2902, 2021-06-10:
Hardware V3: serial number >= 50000, I guess ;)

povlhp, 2021-07-07:
> I would also like to know the new API commands for hardware version 3.
> Is it safe to use hardware version 3 and switch phases over API with any electric vehicle?

I assume you could always stop charging, then switch phases (when we know how to do it), and then start charging again. I would NOT do any on/off switching under load. Will kill any relay/Contactor over time. Scaling up/down is fine.

peterpoetzi, 2021-07-07:
The go-eCharger automatically reduces current, stops charging, and completely disconnects the car from the charger. 

For the car, it feels like you unplug the type-2 cable on the charger, go with the cable to a different charger (that just happens to be 1- or 3-phase) and plug it in on the different charger again.

This is all automated and does not need to be controlled over the API

peterpoetzi, 2021-09-17:
https://github.com/goecharger/go-eCharger-API-v2

closed 2021-09-17 by peterpoetzi