goecharger/go-eCharger-API-v1 #55: Please update API with needed firmware version information

sigma415, 2021-06-01:
Please update API with needed firmware version information, e.g. for parameter "amx" its's not clear, that it does not work in firmware versions lower than 040.

chk2902, 2021-06-04:
you mean V3... yes, there's a new API (API V2... not to confuse anyone... ;) with a different url ("/api/status" and "/api/set") not yet documented. The old one continues to work though.

chk2902, 2021-06-04:
Ach ja, wer Spaß dran hat: `/api/status` liefert z.B. das hier. Ist eine ganze Menge Holz - hoffentlich kann man da per API selektieren, dass man nur "nrg", "alw", "amp", "pha" und ein paar mehr braucht ;)

{
  "stao": null,
  "alw": true,
  "acu": 6,
  "adi": true,
  "dwo": null,
  "atp": null,
  "pwm": 0,
  "sse": "050000",
  "eto": 11230,
  "etop": 11230,
  "wifis": [
    {
      "ssid": "xxx",
      "key": true
    },
    {
      "ssid": "xxx",
      "key": true
    },
    {
      "ssid": "xxx",
      "key": false
    },
    {
      "ssid": "xxx",
      "key": false
    },
    {
      "ssid": "xxx",
      "key": false
    },
    {
      "ssid": "xxx",
      "key": false
    },
    {
      "ssid": "xxx",
      "key": false
    },
    {
      "ssid": "xxx",
      "key": false
    },
    {
      "ssid": "xxx",
      "key": false
    },
    {
      "ssid": "xxx",
      "key": false
    }
  ],
  "scas": 2,
  "scan": [
    {
      "ssid": "xxx",
      "encryptionType": 3,
      "rssi": -31,
      "channel": 1,
      "bssid": "xxx"
    },
    {
      "ssid": "xxx",
      "encryptionType": 3,
      "rssi": -70,
      "channel": 1,
      "bssid": "xxx"
[... 674 lines cut ...]
  "men": false,
  "msb": false,
  "msr": true,
  "dccu": false,
  "data": "{\"i\":30,\"url\":\"https://download.v3.go-e.io/data?token=RnFWHJ9mbOPfMuC2bK5WgcRe0cNg0baPwuhTsTSU7CNzoa16aOAEvHhvb0xCmLXd\"}",
  "dll": "",
  "fck": true,
  "fcc": true,
  "asc": true,
  "lmsc": 10373,
  "lcm": 15,
  "fhi": true,
  "hai": true,
  "hla": true,
  "cdv": null,
  "mscs": 4,
  "msca": 0,
  "mce": false,
  "mcu": "mqtt://test.mosquitto.org:1883",
  "mcs": false,
  "mcc": false,
  "mcca": 0,
  "lri": null,
  "lrr": 0
}