goecharger/go-eCharger-API-v1 #56: Fehler API V1 mit Hardware V3

chk2902, 2021-06-04:
Firmware "050":
Die Einheit von "dws" ist nicht mehr korrekt - ich vermute es sind nun 1/100 kWh, nicht mehr Dekawattsekunden. Ist das korrekt?

chk2902, 2021-06-04:
Firmware 050:
Verzögerte Ausgabe von AMX im Status. Mag Absicht sein, verwirrt aber, wenn man nicht das zurückbekommt, was man gesetzt hat - es aber im anderen Register angezeigt wird:
17:08:22.233: "car":"2","amx":"6","amp":"6","err":"0","ast":"0","alw":"1"
17:08:22.388: http://.../mqtt?payload=amx=13
17:08:23.591: "car":"2","amx":"6","amp":"13","err":"0","ast":"0","alw":"1"
17:08:25.112: "car":"2","amx":"13","amp":"13","err":"0","ast":"0","alw":"1"

chk2902, 2021-06-09:
"nrg[11]" ist nun W (kW/1000) - früher kW/100
"nrg[0]".."nrg[3]" sind nun A, früher A*10

peterpoetzi, 2021-06-09:
Ich kann bestätigen dass das ein Fehler in der Firmware ist, der in der nächsten Version gefixt wird. Dann wird die Einheit wieder wie gehabt Dekawattsekunden sein.

IPSCoyote, 2021-07-03:
Das ist dann leider wieder einmal ein Fehler, besser eine Regression, einer API Version. Fehler passieren. Aber könnt ihr bitte ggf. etwas sorgfältiger dabei sein, wenn ihr neue Firmwares rausbringt? Formatänderungen dürfen nicht sein und auch Kommandos sollten immer noch stabil funktionieren (das kann man doch evtl. einfach mit einer Prüfroutine abprüfen).

Ich habe 2 GO-eCharger und das MQTT?payload=amx=10 funktioniert z.B. in der 0.221 aber nicht in der 0.40. Man behält relativ schwer den Überblick, was ggf. wann dazu gekommen ist. Und wenn's dann noch evtl. anders läuft ist das für Integrationen in z.B. PV-Regelungen ggf. problematisch.

chk2902, 2021-07-23:
Leider ist das in Firmware "051" in der HTTP V1-API immer noch drin, scheint mir. Könnt ihr das mal überprüfen?

Aus dem Live-System: `"dws":"418","dwo":"50",` entspricht etwa 4.18 geladen von 5.0 Soll. Das könnte stimmen.

IPSCoyote, 2021-07-23:
@chk2902 Bei deinen Werten würde ich sagen, der Fehler ist noch drin. DWS müssten *deutlich* größere Werte sein ;)

Blöde Frage: Firmware 050? Firmware 051? Bei meinen Chargern (Hardware Version V2) ist 040 die neueste.... habe ich da was verpasst?

chk2902, 2021-07-24:
"Leider ist das in Firmware "051" in der HTTP V1-API immer noch drin, scheint mir."

Also ab der "050" (die alte Hardware hat's in Dekawattsekunden), immer noch in der "051".

chk2902, 2021-07-24:
Vielleicht sollten sie es einfach drin lassen und das in der Doku reinschreiben (ab Firmware 050...)

IPSCoyote, 2021-07-26:
Ich bin kein Fan davon, bestehende, ausgelieferte Funktionen oder Definitionen (hier DWS) zu verändern. Ich verwende die API in einem Smarthome-Plugin, welches von diversen Usern genutzt wird. Eine solche Änderung, die zumal ggf. unbemerkt passiert, führt zu Regressionen bei den direkten und indirekten Verwendern der API bzw. des Chargers. 

Aus meiner Sicht muss man, will man hier z.B. den Wert in einem anderen Format liefern, einen zweiten Rückgabeparameter in die API einbauen und den alten Wert kompatibel erhalten. 

Trotzdem sollte man natürlich ggf. bei den einzelnen Parametern in der Doku angeben, ab wann sie ggf. verfügbar wurden oder sich etwas verändert hat.

peterpoetzi, 2021-07-26:
In Firmware 051.1 wurde es bereits auf die gewohnten Dekawattsekunden zurück geändert und verhält sich damit wie aktuell beschrieben

Jpsy, 2021-07-27:
Ja, funktioniert wieder.
Besten Dank.

chilobo, 2021-08-25:
Ich hatte an dem binding für openhab gebastelt, um es auch mit der "neuen" GO_E lauffähig zu halten:
[https://github.com/openhab/openhab-addons/pull/10903]
Dort habe ich die Umrechnungsart dann auch hin- und zurückändern müssen.
Wann wird die Beschreibung der Schnittstelle der API V2 wohl zu erwarten sein?

peterpoetzi, 2021-09-17:
Die V2 API ist hier dokumentiert: 
https://github.com/goecharger/go-eCharger-API-v2

closed 2021-09-17 by peterpoetzi