goecharger/go-eCharger-API-v1 #57: API V1 mit Hardware V3

chk2902, 2021-06-04:
verzögerte Ausgabe des "amx"-Registers

17:08:21.169: http://.../status
17:08:22.233: "car":"2","amx":"6","amp":"6","err":"0","ast":"0","alw":"1"
17:08:22.388: http://.../mqtt?payload=amx=13
17:08:23.591: "car":"2","amx":"6","amp":"13","err":"0","ast":"0","alw":"1"
17:08:24.620: http://.../status
17:08:25.112: "car":"2","amx":"13","amp":"13","err":"0","ast":"0","alw":"1"

Auch ist die Antwortzeit der Wallbox relativ lang, allerdings gibt es hier beim Log 2 Apps gleichzeitig, die die Wallbox abfragen - die von go-e und meine.

sebnaf, 2021-08-05:
Ja, und ein weiteres Problem, zB:

http://.../mqtt?payload=alw=1 gibt in der JSON-Response noch den vorherigen, alten Wert zurück.
Das soll bitte behoben werden.

sebnaf, 2021-08-05:
Unsere Tests haben ergeben, dass die Änderung erst nach 7 Sekunden ersichtlich ist.

peterpoetzi, 2021-09-17:
Mit V3 Firmware 051.3 und höher werden gesetzte Parameter schneller übernommen

closed 2021-09-17 by peterpoetzi

chk2902, 2021-09-17:
(das war die SW 51.4...)