goecharger/go-eCharger-API-v1 #63: check if new firmware available

clyso, 2021-06-18:
is there any possibility to check for new firmware updates besides using the official app while connected via cloud?
I would like to implement a check in my home assistant that i get notified when a new update for the charger is available.

sigma415, 2021-07-16:
I also would really appreciate such a firmware check feature. I hab to downgrade on V033 (HW V2) because 040 killed the restart function of my box. And iOS app only tells me, that there is some new FW version available, but not which one .... :(

peterpoetzi, 2021-09-17:
The restart function is fixed with V2 Firmware 041

closed 2021-09-17 by peterpoetzi

peterpoetzi, 2021-09-17:
The V2 does not know the newest firmware as an API parameter, but you can read it in the changelog in the app