goecharger/go-eCharger-API-v1 #64: <ip>/status -> couldn't take api lock in 3s, please try again later

chk2902, 2021-07-03:
Wenn ich per Browser ("http://192.168.2.59/status" oder "http://192.168.2.59/api/status") auf die Box zugreifen will (API V1 und V2, oder auch nur V2 enabled), bekomme ich bei der neuen Hardware als Antwort:

"couldn't take api lock in 3s, please try again later"

Per Volley (Android-Lib) bekomme ich Server Fehler 500 gemeldet.

Früher (TM) hat die Box auf "/status" und "/api/status" geantwortet - Firmware "050". Jetzt habe ich hier eine "050.1", und der Zugriff klappt nicht.

chk2902, 2021-07-03:
OK, erst hatte das Update nicht geklappt, jetzt beim zweiten Anlauf schon. Mit der 50.2 geht es wieder.

chk2902, 2021-07-04:
Heute tritt es wieder auf. "http://192.168.2.59/status" "couldn't take api lock in 3s, please try again later"

API V2 enabled, V1 disabled.
"http://192.168.2.59/status" -> "This URI does not exist" -> das ist OK

API V2 disabled, V1 enabled.
"http://192.168.2.59/apistatus" -> "This URI does not exist" -> das ist OK
"http://192.168.2.59/status" -> "http://192.168.2.59/status" "couldn't take api lock in 3s, please try again later" -> Fehler

V1 und V2 enabled.
beide URLs ergeben "couldn't take api lock in 3s, please try again later"

Neustart -> beide APIs gehen.

Das Teil ist instabil.

miurlit, 2021-07-09:
Habe das gleiche Problem, bzw. was ich weiter eingrenzen konnte:
- nach einem Neustart funktioniert die Statusabfrage ohne ladendes Fahrzeug (zumindest über 1 Stunde getestet)
- wenn das gleiche mit einem Fahrzeug dran gemacht wird, funktioniert die API nur wenige Sekunden (5-20sec) dann erhalte ich die `couldn't take api lock in 3s`, please try again later"
-> habe das aber nur mit der V1 API getestet, da ich V2 bisher nicht in Verwendung hatte. Außerdem tritt das nur bei meiner neuen (Hardware V3) auf, bei meiner alten V2 war das nie ein Problem.

sslneve, 2021-07-11:
Ich habe exakt das selbe Problem, zusätzlich noch beim Dauerping festgestellt, dass es viele langsame Antworten (>600ms) gibt immer wieder und auch ein paar Packet Losses. Mein alter go-e Charger hängt am selben Hotspot und verhält sich sowohl bzgl API als auch Ping einwandfrei. Liebe Leute, bitte fixed das schnellstmöglich, ich mache da jetzt schon 3 Wochen rum mit und ärgere mich, weil ich den Charger nicht nutzen kann mit meiner openWB Steuerung.

jowo1217, 2021-07-13:
Auch ich habe das gleiche Problem. Es tritt leider sporadisch auf und ich kann kein pattern erkennen. Ich nutze noch die alte API v1 aufgrund der fehlenden Doku für v2.
Hatte vorher den alten go-e und da lief es wie geschnitten Brot. Da ich seit Kurzem PV-Überschuss laden mache, bin ich auf eine verlässliche Steuerung der WB angewiesen.
BITTE, BITTE schnellstens fixen.

sslneve, 2021-07-13:
Wie ich gerade erfahren habe, ist Herr Pötzi von go-e diese Woche im Urlaub. Das wird also leider nichts mit "schnell"... Ich verstehe nur nicht, wie man so unausgereiften und offenbar mindestens schlecht getesteten Kram auf den Markt werfen kann.

chk2902, 2021-07-13:
Er hat mir vor ein paar Tagen geschrieben, dass sie dieses Problem gefixt hätten und die neue FW in internen Tests prüfen.

Ich könnte mir vorstellen, dass die API serialisiert wunderschön läuft, aber wenn da API-Aurufe "verschachtelt" sind oder zu schnell aufeinander folgen, dass der Fehler dann irgendwann auftritt. Da sie selbst das so nicht nutzen (WebSocket-API), fällt es ihnen in "einfachen" Tests nicht auf.

Dann müssten sie rechtzeitig Beta-FWs rausgeben für externe Tester für Home- oder andere Automation. Mein go-e 5xxxx liegt jedenfalls in der Ecke und kann nicht genutzt werden. Ich habe mich nun in der Zwischenzeit mit einer SmartWB angefreundet, die ich vom Hersteller auch noch als Dauerleihgabe bekommen habe - *das* ist ein Service für mich, der ich den Wallbox-Anbietern auch einen deutlichen Mehrwert biete.

BEi go-e dagegen Schweigen im Walde. Keine Test-Hardware, keine Test-Software, keine Doku. Nicht mal Beta-Doku und Beta-Firmware.

chk2902, 2021-07-23:
Mit der fwv 51 scheint es zu laufen.

closed 2021-07-23 by chk2902