goecharger/go-eCharger-API-v1 #66: Some findings about the api of the hardware version 3

ddrrhh, 2021-07-05:
## HotSpot

The new hardware establishs a hotspot in a new ip-adress range.
The hotspot of our charger is found at adress 10.128.250.281

With 
```
http://10.128.250.181/status 
```

the status is requested.

The new status json object is returned.

## API features
Parameter can be set by:

```
http://*.*.*.*/api/set?parameter=value
```
The charger returns a message of the format:
```
x succeeded and x failed
```
Setting multiple parameter at once with the obvious syntax
```
http://*.*.*.*/api/set?parameter=value&parameter2=value2
```
seems not to work. Only the first parameter is respected.

### Switching of phases
One of the most interesting features of the new hardware is the possibility to 
switch between one and three phases. This enables one to charge the car with lower excess from the 
pv. The switching causes, at least the our car, to reinit a new charging session. 
For this reason there is a time gap after phase switch and restarting the charging prozess.

Two parameters of the new status object change after switching between one and three phases using the APP.

For three phases one finds:
* "fsp":false  
* "psm":2

For one phase:
* "fsp":true
* "psm":1

Calling:
```
http://*.*.*.*/api/set?fsp=true
```
returns:
```
0 succeeded and 2 failed
```

Calling:
```
http://*.*.*.*/api/set?psm=1
```
returns:
```
1 succeeded and 0 failed.
```

The APP shows a switch to one phase.

### Setting the current

In the new api the parameter **amx** has gone. This corresponds to the finding that every change of the current seems to change always both parameter **amx** and **amp** in the old api status. 

The parameter **amp** seems to be the candidate for dynamical changing of the current in the new api.

peterpoetzi, 2021-07-05:
Your findings are correct. I still owe the community a full api documentation for V3 chargers, we are working on it

chk2902, 2021-07-22:
remember to enable HTTP API V2 for the "/api" path.

kaeferfreund, 2021-07-28:
Any progress on the documentation?

0xFEEDC0DE64, 2021-09-17:
https://github.com/goecharger/go-eCharger-API-v2

closed 2021-09-17 by peterpoetzi