goecharger/go-eCharger-API-v1 #69: Fehler API V1 mit Hardware V3:

chk2902, 2021-07-23:
Leider wird die Lademenge pro Ladung nicht mehr dem ersten Benutzer zugeordnet  (mindestens in Firmware 51, vorher habe ich das überhaupt nicht überprüft), sondern nur zur Gesamtmenge addiert.

Früher (Hardware V1) wurde die Ladung, die per API gestartet wurde, immer dem ersten Nutzer zugeordnet. Jetzt ist alles leer trotz 5 kW Ladung:

`"eca":"0","ecr":"0","ecd":"0","ec4":"0","ec5":"0","ec6":"0","ec7":"0","ec8":"0","ec9":"0","ec1":"0"`

0xFEEDC0DE64, 2021-09-17:
wurde eine karte als benutzer 1 angelernt und vor dem ladevorgang zum lesegerät hingehalten?

tobintax, 2022-08-31:
> wurde eine karte als benutzer 1 angelernt und vor dem ladevorgang zum lesegerät hingehalten?

Er schreibt ja dass die Lademenge früher ohne hinhalten der RFID Karte immer dem ersten Benutzer zugeordnet wurde. Unabhängig davon ob man eine Karte ranhält oder nicht. Nun ist das nicht mehr so. Wenn keine RFID Authorisierung stattfindet wird die Lademenge zwar bei der Gesamtmenge gezählt, aber nicht mehr dem ersten RFID Nutzer mehr zugeordnet. Sieht für mich so aus als ob es hier eine Änderung im Handling diesbezüglich gab.