goecharger/go-eCharger-API-v1 #70: HTTP API V1 mit FW 051

DaveMe81, 2021-07-24:
Hallo!
Habe heute das Update auf die Firmware 051 (Hardware V3) gemacht. Im Changelog steht, dass die API stabiler sind.
Aktuell erfahre ich aber genau das Gegenteil. Ein einfacher Aufruf (bei mir aller 2min) des Status ("x.x.x.x/status") für ein timeout (nach der 3. spätestens nach der 6. Abfrage). Erst ein Neustart initiiert in der App schafft temporär Abhilfe.

CHulkW, 2021-07-24:
Hallo, bei mir ist es das gleiche Problem

dataeasy74, 2021-07-24:
Gleiches Problem bei mir gefühlt stützt die API V1 noch schneller ab als vorher ! Somit ist aktuell kein PV geführtes Laden möglich kann ich igrndwie downgraden auf die 048 Version die lief viel stabiler :-(

DaveMe81, 2021-07-24:
> Gleiches Problem bei mir gefühlt stützt die API V1 noch schneller ab als vorher ! Somit ist aktuell kein PV geführtes Laden möglich kann ich igrndwie downgraden auf die 048 Version die lief viel stabiler :-(

Genau so ist es. Und das zu dieser Jahreszeit. Habe mir ein schönes Script geschrieben, dass meine Ansprüche erfüllt, um den Eigenverbrauch zu erhöhen. Aber nein, go-e stürzt ab. Ich ärgere mich extrem.

Jpsy, 2021-07-25:
Ich hatte genau das gleiche Problem. Timeouts bei Zugriff auf /status und /api/status nach Update auf 051. Nach jedem Neustart des Chargers per App waren die Endpunkte immer für einige Minuten zu erreichen. Dann liefen die Requests wieder in Timeouts. 

Geholfen hat mir ein Tipp im OpenWB-Forum: 
Den go-eCharger kurz komplett stromlos machen. 
Seitdem läuft die API unter 051 bei mir wieder stabil. Muss man erstmal drauf kommen, wenn die App einem einen "Neustart" anbietet, der nicht hilft.

DaveMe81, 2021-07-25:
Danke für den Tipp, werde ich jetzt nochmal versuchen. Ich war der Meinung, dass ich das gemacht hatte. Ich schalte ihn mal komplett stromlos für einige Minuten und stecke auch das Kabel ab. Mal sehen

DaveMe81, 2021-07-25:
Negativ! Nach 25min wieder das Problem mit Timeout. Also auch ein stromlos schalten hat nichts gebracht

Jpsy, 2021-07-25:
Das ist seltsam. Bei mir läuft der go-e seit dem Kaltstart jetzt seit über 30 Stunden völlig stabil. Ich kann das gut überprüfen, weil die Hausautomatisierung die API alle 20 Sekunden pollt.

DanDan800, 2021-07-25:
Moin bei mir tritt das Problem auch auf. Api ist nach kurzer Zeit 3 min nicht mehr erreichbar. Ist erst nach den letzten Updates aufgetreten. Neustart bringt nur kurze Besserung. 

Gruß Daniel 

(2x go echarger v3 ;firmware 051)

DaveMe81, 2021-07-25:
> Das ist seltsam. Bei mir läuft der go-e seit dem Kaltstart jetzt seit über 30 Stunden völlig stabil. Ich kann das gut überprüfen, weil die Hausautomatisierung die API alle 20 Sekunden pollt.
> Im Diagramm der Spannung auf L1 kann man gut den Zeitpunkt des Updates erkennen (Beginn der Flatline), dann die Versuche, per App zu resetten (einzelne Änderungen in der Flatline) und dann die "Reparatur" durch Kaltstart.
> 
> [image]

Ich glaub dir das, kein Problem. Ich verstehe es ja auch nicht. Habe bei mir heute mal echt alles angeschaltet bzgl. laufender Automatisierungen. Sogar mal ein separates WLAN nur mit dem go-e und die Polling-Zeit verlängert oder verkürzt. Nix, irgenwann kommt immer ein timeout.

dataeasy74, 2021-07-25:
> Ich hatte genau das gleiche Problem. Timeouts bei Zugriff auf /status und /api/status nach Update auf 051. Nach jedem Neustart des Chargers per App waren die Endpunkte immer für einige Minuten zu [...]

NO bringt bei mir leider garnichts der GoE V3 hängt sich weg mal früher mal später... ein stabiles Regeln durch externe Steuerung via API ist nicht möglich... ich bin auch echt genervt.... hatte erst vor ein paar Wochen meinen alten GoE V2 mit der FW 048 gegen einen V3 getauscht bekommen weil der V2 Probleme beim PV geführten laden gemacht hat bei meinem SEAT Mii ( beim Kona klappe es immer ohne Probleme... mit dem V3 wurden die Probleme leider noch viel schlimmer ...

man10to, 2021-07-25:
[image]

Hallo,
Ich hab hier 3x die v2 mit 0.40 und 3x die v3 mit 0.51 und frage die alle per iobroker ab..
Heute mit unserem Kia Niro Phev wieder gekommen und ihn mit seiner 9kwh Batterie zum laden an nr. 6 angeschlossen..
Hat sich über 800 kWh geholt ;-)
Also ich wegfuhr war noch die 0.50 drauf, hab dann über die Ferne die 0.51 drauf gemacht ...
Mal 1-2 frage.
1. bekommen die das wohl wieder hin ?
2. kann man dann die Verbräuche wieder zurücksetzten ?
3. warum machen die sowas ? 

Ein recht verwunderter Maschinenbauer der solche Einblicke in die Elektrotechnik und Informatik nicht haben wollte ;-)

Unsere SPS Dinger laufen auch nicht fehlerfrei, aber sowas ist schon krass....

peterpoetzi, 2021-07-25:
Das Problem wurde mit 051.1 gefixt (ist bereits für alle V3 verfügbar)

@man10to um welchen Verbrauch geht es?

sslneve, 2021-07-25:
@peterpoetzi Ich würde ja gerne updaten, aber seit Freitag sehe ich nur "Update-Server nicht erreichbar", egal ob ich direkt auf dem Hotspot oder über mein Haus-WLAN drauf bin, hab auch schon den Charger mit dem AdHoc AP meines iPhone verbunden. Er ist nicht zum Update zu bewegen.

sslneve, 2021-07-25:
Nebenbei: jetzt gerade komme ich per Haus-WLAN gar nicht mehr drauf. Timeout. Der alte v2 Charger lässt sich weiterhin problemlos ansteuern. Was für ein Dreck.

DanDan800, 2021-07-25:
> @peterpoetzi Ich würde ja gerne updaten, aber seit Freitag sehe ich nur "Update-Server nicht erreichbar", egal ob ich direkt auf dem Hotspot oder über mein Haus-WLAN drauf bin, hab auch schon den Charger mit dem AdHoc AP meines iPhone verbunden. Er ist nicht zum Update zu bewegen.

Moin, update hat bei mir gerade funktioniert. Ob das Problem behoben ist, wird sich wohl bald zeigen..

peterpoetzi, 2021-07-25:
@sslneve Wenn die API stecken geblieben ist, geht auch die App Verbindung im lokalen Netzwerk nicht. Update Server nicht erreichbar wird dann angezeigt, wenn keine Cloud Verbindung besteht, z.B. weil diese in den Erweiterten Einstellungen deaktiviert ist

sslneve, 2021-07-25:
@peterpoetzi Cloud Verbindung ist natürlich an, ist eh default. Ich habe gerade nochmal alles probiert, samt Zurücksetzen und Strom weg. Immer noch das selbe. Gibt es bekannte Probleme in einem WLAN aus Ubiquiti Access Points gesteuert von einem Unifi Controller?

peterpoetzi, 2021-07-25:
@sslneve hmm nein, ich habe bei mir an allen Standorten ebenfalls nur Unifi APs. Um welche Seriennummer handelt es sich denn, eventuell sehe ich etwas dazu im Error Log vom Load Balancer (gerne auch per email an office (at) go-e.co)

sslneve, 2021-07-25:
@peterpoetzi 060527

peterpoetzi, 2021-07-25:
Danke, Problem ist behoben. Sollte jetzt verfügbar sein

sslneve, 2021-07-25:
Stimmt, danke! Was klemmte da denn?

peterpoetzi, 2021-07-25:
Seriennummern ab 60.000 waren noch nicht für das Update freigeschalten

sslneve, 2021-07-25:
Na prima.... Hoffen wir, dass das Gerät nun endlich, endlich tut. Danke!

sslneve, 2021-07-25:
Ok, die Fernsteuerung von der openWB aus über die v1 API funktioniert jetzt auf den ersten Blick. Im Gegensatz zum v2 Charger musste ich allerdings zuerst einmal in der App auf den Button tippen zum Autorisieren.

man10to, 2021-07-25:
Die nr 6 mit den 870 kWh

DaveMe81, 2021-07-26:
> Das Problem wurde mit 051.1 gefixt (ist bereits für alle V3 verfügbar)
> 

Neuer Tag - neue Firmware!

Das ging fix - super! Ich probiere es gleich mal aus.

dataeasy74, 2021-07-26:
THX FW 051.1installiert Feedback folgt sobald wieder Strom ins Auto muss ;-)

DanDan800, 2021-07-26:
..bei mir läufts wieder wie gewohnt stabil mit der neuen FW.
@peterpoetzi 👍 Super Job! ...und das auf einen Sonntag Abend

DaveMe81, 2021-07-26:
> 
> Das ging fix - super! Ich probiere es gleich mal aus.

Seit 1,5h keine timeout-Meldungen mehr! Sehr gut! @peterpoetzi TOP!!! Vielen Dank!

dataeasy74, 2021-07-26:
bis dato läuft es (habe ein Testscript im IOBroker gebaut was einfach Werte abfragt und schreibt) THX @peterpoetzi wo lag es denn dran ;-)

closed 2021-07-26 by peterpoetzi