goecharger/go-eCharger-API-v1 #75: JSON Response does not reflect changed value

sebnaf, 2021-08-05:
Hardware: V3

Changing a value like `http://.../mqtt?payload=alw=1`does still return the old, original value in the JSON Response.

This should be changed to reflect the freshly set value.

Thank you

sebnaf, 2021-08-05:
according to the documentation, this really seems to be a bug.

4. Rückgabewerte :
Lokales WLAN / Hotspot
Verbindung	Rückgabe
WLAN Hotspot	Komplettes status JSON Objekt mit bereits geänderten Wert
WLAN lokales
Netzwerk	**Komplettes status JSON Objekt mit bereits geänderten Wert**
Bei jedem Status Request und jedem Kommando wird das Status JSON-Objekt
zurückgegeben. Ein nicht erfolgreiches Kommando erkennt man daran dass sich der Wert im
Status Objekt nicht geändert hat.

sebnaf, 2021-08-11:
Any remarks here? Thank you

Optic00, 2021-08-31:
+1 just ordered a v3 for a relative

peterpoetzi, 2021-08-31:
This bug is fixed internally, we will release an firmware update very soon

sebnaf, 2021-08-31:
Excellent news, thank you @peterpoetzi!

amnesia106, 2021-09-05:
[image]

sebnaf, 2021-09-05:
@peterpoetzi Ich lasst hier einen Response 13s warten - das ist doch nicht wirklich elegant… Könnt ihr da bitte nochmals nachbessern?

 sna@arbeitsgeraet  ~  time curl 'http://10.10.20.22/mqtt?payload=alw=1'

{"version":"B", ... ,"loa":0,"lch":507}
curl 'http://10.10.20.22/mqtt?payload=alw=1'  
0.00s user 0.00s system 0% cpu 13.417 total

sebnaf, 2021-09-05:
@peterpoetzi Falls dieses Problem mit der API v2 nicht bestehen sollte, bitte ich umgehend um die Dokumentation, damit wir das umbauen können. Die 13s Wartezeit sind architektonisch noch schlimmer als der Fehler an sich.

CHulkW, 2021-09-06:
+1

Da konnte ich eher mit dem Fehler leben, als mit der langen Wartezeit. So ist das leider wirklich nicht gut nutzbar.

sebnaf, 2021-09-06:
Die ersten User melden einen Timeout. 
Es wäre wirklich gut zu wissen, ob das mit der API v2 besser funktioniert. Hat irgendjemand Zugang dazu oder kann sich @peterpoetzi bitte dazu äussern? Danke

chilobo, 2021-09-06:
Auch bei mir wird die GO-E unter openhab nicht mehr initialisiert:
`Total timeout 5000 ms elapsed`
IP/status und IP/api/status liefern im Browser noch Daten, dauert ca. 7 s

peterpoetzi, 2021-09-06:
Ich kann bestätigen dass bei manchen Usern die v1 API jetzt mehrere Sekunden zum Antworten braucht. Wir arbeiten hier an einem Bugfix

sebnaf, 2021-09-06:
@peterpoetzi Danke. Da nun doch einige Implementationen seitdem ein Timeout liefern - mit welchem Zeithorizont dürfen wir hier rechnen?

chilobo, 2021-09-07:
In Openhab habe ich die timeouts beim Lesen und Senden auf 10 s hochgesetzt. Jetzt klappt die Steuerung wieder.

sebnaf, 2021-09-07:
@chilobo Das wird wahrscheinlich nur mittelfristig etwas bringen, da das Timeout länger wird, je länger die Uptime ist. 
Hoffen wir auf einen baldigen Fix.

chilobo, 2021-09-07:
Scheint zu stimmen. Warum wird das Timeout länger? 
Macht es dann Sinn:
- die Box ab und an im Sicherungskasten stromlos zu machen
- die Ladung neu zu starten 
- oder reicht es, die Box aus- und wieder anzuschalten? 

Habe gerade mal Letzteres probiert, scheint nicht sicher zu helfen.

sebnaf, 2021-09-08:
Hallo @peterpoetzi, könnten wir bitte eine Aussage betreffend Zeithorizont Bugfix bekommen? Danke

peterpoetzi, 2021-09-08:
Es handelt sich hier um eine Thread-Locking Problem, das bereits intern gelöst und und gerade getestet wird. Ein Update erwarte ich für heute 09.09

sebnaf, 2021-09-09:
Danke für das Update. Das sind sehr erfreuliche News. Danke, dass ihr so schnell seid.

closed 2021-09-11 by peterpoetzi