goecharger/go-eCharger-API-v1 #76: v3 - Where to get the auth. token from?

eicosan, 2021-08-05:
Just received my new e-go charger v3. 
but where do I get the cloud token from. Everywhere it says it is on the reset card, but it’s not!! 

Thanks for help! 
eicosan

sebnaf, 2021-08-05:
I had exactly the same conversation with go-e support yesterday.
There is no cloud token with v3.

>der Cloud-Token, wie dieser von der älteren Generation bekannt war, existiert nicht mehr in seiner alten Form. 

Stattdessen hinterlegen Sie nun ein eigenes Passwort im Reiter Einstellungen unter Passwort. Sollten Sie das Setup mit dem Assistenten durchgeführt haben, erstellen Sie im letzten Schritt das Passwort.
Das dort erstellte Passwort gilt als Ihr Cloud-Token, wobei dieser mit Apps von Drittanbietern nicht kompatibel ist. 

Apps von Drittanbietern nutzen noch die API v1 und auch deren Zugriffsarten, welche allerdings für go-eCharger der dritten Generation nicht kompatibel sind. 
Erst wenn wir unsere neue API-Dokumentation fertigstellen und als Download zur Verfügung stellen, können Apps von Drittanbietern nachziehen und deren Apps anpassen.

eicosan, 2021-08-05:
> I had exactly the same conversation with go-e support yesterday.
> There is no cloud token with v3 yet.
> 
> > der Cloud-Token, wie dieser von der älteren Generation bekannt war, existiert nicht mehr in seiner alten Form.
> 
> > Stattdessen hinterlegen Sie nun ein eigenes Passwort im Reiter Einstellungen unter Passwort. Sollten Sie das Setup mit dem Assistenten durchgeführt haben, erstellen Sie im letzten Schritt das Passwort.
> > Das dort erstellte Passwort gilt als Ihr Cloud-Token, wobei dieser mit Apps von Drittanbietern nicht kompatibel ist.
> 
> > Apps von Drittanbietern nutzen noch die API v1 und auch deren Zugriffsarten, welche allerdings für go-eCharger der dritten Generation nicht kompatibel sind.
> > Erst wenn wir unsere neue API-Dokumentation fertigstellen und als Download zur Verfügung stellen, können Apps von Drittanbietern nachziehen und deren Apps anpassen.

Danke sebnaf. Das ist eine sehr ernüchternde Antwort. Ehrlich gesagt verstehe ich es nicht wie man etwas verkaufen und bewerben kann was nicht dokumentiert ist. Außerdem habe ich bisher immer verstanden, dass die API v1 schon noch funktioniert nur halt nicht die neuen Features beinhaltet.

peterpoetzi, 2021-09-17:
Die v2 API ist mittlerweile dokumentiert, jedoch wird von den neuen go-eChargern V3 auch noch die v1 API wie bisher unterstützt. Für die neuen Funktionen der V3 ist es jedoch notwendig die v2 API zu verwenden.

closed 2021-09-17 by peterpoetzi

Newan, 2021-09-17:
Wo findet man den die docu 2.0 und wie kann man sich authen ohne token?