goecharger/go-eCharger-API-v1 #77: Unable to set MQTT URL (Hardware v3)

eicosan, 2021-08-07:
After activating the option "Aktiviere MQTT API" I can change the URL below, but after saving the configuration it returns to the default one: mqtt://test.mosquitto.org:1883.

DaveMe81, 2021-08-08:
Same here! I was hoping that this is already known an to be fixed soon.

sebnaf, 2021-08-08:
Same behaviour here as well.

Marv2190, 2021-08-08:
Tested, just now and works for me.

eicosan, 2021-08-08:
I just figured out that setting the MQTT server is possible using the new REST API:
http://##GO-E_IP##/api/set?mcu="mqtt://##MQTT_HOST##:1833"

EugenGanshorn, 2021-08-31:
I can confirm, there are no status updates published to the topic "go-eCharger/##SERIAL_NR##/status"

is this intentional, or when will it be fixed?

peterpoetzi, 2021-09-17:
The V3 has a slightly different MQTT protocol which sends all API parameter to separate topics: 
https://github.com/goecharger/go-eCharger-API-v2

closed 2021-09-17 by peterpoetzi