goecharger/go-eCharger-API-v1 #79: V3 - FW51.1 - Modbus Register 299

JoGi65, 2021-08-16:
Hallo,

ist es normal, wenn man per Modbus Register 299 schreibt, dass sich Register 300 immer mitverändert?

LG Johannes

Sillium007, 2021-09-02:
Das wäre durchaus bedenklich. Klingt aber schon sehr danach wenn der Wert nach dem Ausschalten des Chargers beim nächsten Start immer noch da ist.

JoGi65, 2021-09-02:
Ich habe vor einigen Tagen auf meine Email Anfrage eine Antwort erhalten.

**Zitat gekürzt:**
_Sehr geehrter Herr 

ich habe nun eine Rückmeldung aus unserer Entwicklung erhalten. 
Ihr Charger entspricht der neuesten Generation, weshalb die Beschränkung von ca. 100.000 Schreibvorgängen nicht mehr aktuell ist. 
Die neue Generation kann (quasi) unlimitiert oft schreiben und persistiert daher den Strom immer - das ist also ein gewolltes Verhalten._ 

Ich hoffe das dem so ist, auch wenn es für mich etwas merkwürdig ist, da sich 300 ja praktisch erübrigt.

closed 2021-09-17 by peterpoetzi