goecharger/go-eCharger-API-v1 #83: incompatibility http method API V1 with Hardware V3

romor, 2021-09-02:
Up to now, we used an HTTP GET request to read the status information and an HTTP POST request to change parameters, e.g. by
`curl -X POST http://<ip>/mqtt?payload=amp=13`

This worked fine with old hardware up to firmware 0.40. However, with a V3 device with firmware 051.1 this yields an error:

`Request method for this URI is not handled by server`

I guess we somehow used the API incorrectly up to now. However, I propose to also allow POST requests for changing parameters on the new hardware / firmware to keep backwards compatibility (and to follow common practices using HTTP methods).

chilobo, 2021-09-03:
In
[https://github.com/openhab/openhab-addons/issues/10791#issuecomment-864535124](url)
I proposed a change to change the goe-binding for openhab:
I had to change the method to submit the mqtt-request from POST to GET.

peterpoetzi, 2021-09-17:
hardware-V3 Firmware 051.5 and later also allows PUT and POST requests in addition to GET, which then behaves similar to the hardware-v2

closed 2021-09-17 by peterpoetzi