goecharger/go-eCharger-API-v1 #84: Falsche gemessene Stromstärken und Leistungen nach Wechsel der Sollstromstärke (Hardware V3, HTTP-API V1)

chk2902, 2021-09-04:
Gerade hat ein Kunde eine 3P-Ladung durchgeführt und wundert sich über das Zackenmuster.

Man sieht im Log schön, dass nrg[4..6] die falschen Werte anzeigen:

16:49:44.026: [240,242,240,0,39,27,28,6,4,4,0,150,66,66,61,11]
"... amx=6"
16:49:44.445: [241,242,240,0,10,10,10,0,0,0,0,7,11,10,8,11]

-> Zur fast gleichen Zeit (16:49:44.026 und 16:49:44.445) meldet die Wallbox völlig verschiedenen Stromstärken (Erst 3.9, 2.7, 2.8A, dann 1.0, 1.0, 1.0A)!

Kurz (naja) danach dann sieht es wieder richtig aus, bis dahin bleibt es 1.0:

16:49:52.936: [240,241,240,0,51,46,44,11,10,9,0,319,93,93,91,9]

Das passiert immer wieder mal für ein paar Sekunden nach dem Ändern der Stromstärken:

"... amx=7"
17:00:48.595: [240,242,239,0,10,10,10,0,0,0,0,8,12,8,9,1]
...
17:00:57.028: [241,239,239,0,59,54,52,13,12,11,0,379,95,95,93,2]

Das ist natürlich nicht der feinen Wallbox Art!

Man beachte auch, dass die Leistungen auf 0 sind bei Stromstärke 1A... ?!

chk2902, 2021-09-04:
Ergänzung: das passiert bei mir nicht - der Kunde hat aber einen Wagen für das "Aufwecken" eingestellt, vielleicht hat das damit zu tun?

chk2902, 2021-09-05:
Noch eine Ergänzung: das scheint nicht immer der Fall zu sein. Beim gleichen Kunde heute mit 1P-Ladung trat das Problem nicht auf.

peterpoetzi, 2021-09-17:
Ich denke hier handelt es sich nicht um einen Bug in der Ladebox sondern einfach um ein sonderbares Verhalten des Autos. Das kann den Strom ja selbst regeln bis zu dem vorgegebenen Maximalwert

closed 2021-09-17 by peterpoetzi

chk2902, 2021-09-17:
Strom auf 3*1 A, aber Ladeleistung 0W? Wie soll das das Auto machen können???

peterpoetzi, 2021-09-17:
Wenn der Leistungsfaktor fast 0 ist, die 1A entstehen durch den Blindstrom der EMV Kondensatoren im Lader wenn er keinen Strom zieht aber trotzdem aktiv ist (was z.B. beim aufwärmen sein kann)

chk2902, 2021-10-01:
Spannend... Danke.