goecharger/go-eCharger-API-v1 #85: api v2: status

sebnaf, 2021-09-06:
Good day,

analyzing the JSON gathered from http://{{goeUrl}}/api/status, I do see over 600 lines (!) of nearby WLAN networks.

I'd love to to specify an array of values to be sent back only - we do not want to send and proceed not needed entries.

Thank you

chk2902, 2021-09-10:
I don't know why the API2 is there anyway... 

but you can still use API1 and be happy - where's the advantage of API2? A bit more structured, maybe more precise values (we don't know yet here) but that's it.

sebnaf, 2021-09-10:
I assume API v2 has been created to satisfy the go-e cloud services. Most of the versioned APIs have deprecated the v1/older versions sooner or later.

In this case I want to use slim and fast API response bodies.

This brings up another interesting topic: are my nearby WLANs sent back to the go-e cloud as well?

chk2902, 2021-09-10:
I would guess that's not correct - the go-e app uses WebSockets to communicate with the servers and the device. I have started WebSocket sniffing weeks ago but gave up - no documented authentication (and a very bad implementation if you need to use standard libraries due to thousands of iterations - absolute overkill and taking many seconds to calculate even on a fast smartphone). In principle this is much better (sends delta information as push messages), thus this would have been preferred by me, but then, they lack of documentation and reliably working wallboxes anyway.

sebnaf, 2021-09-10:
I do not want to continue offtopic any further, thank you.

chk2902, 2021-09-10:
Well that's OK but you asked whether the Wifi information is sent to the server ("are my nearby WLANs sent back to the go-e cloud as well?") - and I did answer that as far as I remember, this is the case.

So please tell me why this is off-topic.

sebnaf, 2021-09-10:
@chk2902 can you assure that the WLANs are not sent to go-e in any case or is this just an assumption doing some sniffing?

chk2902, 2021-09-11:
@sebnaf Just an assumption, yes. The communication between the wallbox and the go-e servers must be captured on the router, and I did not do this.

chk2902, 2021-09-12:
Your wish was fulfilled...
https://github.com/goecharger/go-eCharger-API-v2 : you can set a filter of registers to read.

sebnaf, 2021-09-13:
awesome.

closed 2021-09-13 by sebnaf