goecharger/go-eCharger-API-v1 #88: V3 Mqtt: go-eCharger/0xxxxx/status ist null

rompriest, 2021-09-20:
Ich habe seit 2 Wochen 2x go-charger V3 installiert und habe mich jetzt an die Einbindung in node red gemacht.

dazu habe ich existierende Flows importiert und angepasst. diese flows basieren auf mqtt und die basis zum einlesen und befehle abgeben sind die felder go-eCharger/0xxxxx/status bzw. go-eCharger/0xxxxx/cmd/req

Mein lokaler mqtt server und auch mqtt explorer zeigen aber, dass auf go-eCharger/0xxxxx/status keine Informationen durch die go-e gepostet werden, Wert bleibt null. Ich habe sämtliche Optionen in der app bezüglich V1 api, V2 api usw. probiert, ohne Erfolg. Wenn ich …/status manuell befülle, funktioniert der flow übrigens

was übersehe ich?

JochenFriedrich, 2021-09-21:
V3 nutzt eine komplett andere MQTT Struktur.

Siehe:
https://github.com/goecharger/go-eCharger-API-v2/blob/main/mqtt-de.md

rompriest, 2021-09-22:
Ja, das ist leider korrekt
Wobei im http Protokoll der status ja weiterhin über http://x.x.x.x/status abgreifbar ist

Insofern nutze ich jetzt die http Schnittstelle um den aktuellen Status abzugreifen. Befehle setze ich im mqtt Protokoll über /go-eCharger/0xxxxx/alw/set ab.

**Da habe ich ein neues Problem gehabt**: alw ist nicht beschreibar, weder über 0/1 noch über true/false. Fehlermeldung ist "...tried to set api key without setter (false)". 
Alternativ nutze ich jetzt frc. Da kommt kurioserweise die gleiche Fehlermeldung, die Änderung wird hier aber trotzdem übernommen

die anderen Register die für mich fürs PV-gestützte Laden wichtig sind, wie amp, stp etc. funktionieren übrigens ohne Fehlermeldung

FW 051.4

DerZiad, 2023-03-08:
@rompriest , bei dem Attribut "alw" habe ich auch den gleichen Fehler.