goecharger/go-eCharger-API-v1 #89: Charger V2 stürzt weiterhin ab! 🤷🏻‍♂️

kaeferfreund, 2021-09-25:
Hallo zusammen,

meine beiden am häufigsten genutzten go-e V2 stürzen auch mit der "neuen" 041.0 Firmware ab. Diesmal sogar schlimmer als vorher (vorher hat der Charger in 5/10 Fällen wenigstens weiter geladen):

Der Charger fängt an zu laden, läd wenige Sekunden, dann wird plötzlich mehrfach die Kabel-Verriegelung betätigt im Charger während die LEDs vom Charger und das Auto einen Ladevorgang signalisieren. In der App wird erst der korrekte, dann plötzlich ein sehr geringer Ladestrom bis runter auf 10W angezeigt, geht dann wieder hoch auf den Zielwert, wieder runter bis das Auto dann keine Lust mehr hat und in den Fehlerzustand geht.

Das Spiel ließ sich mit 2x MiniCooper SE und 1x ID4 beliebig oft wiederholen. Ein Neustart in der App hat das Problem ebenfalls nicht behoben.

Alle Autos laden an meinen V3 Boxen, einen Parkplatz weiter einwandfrei.

Nach dem ich die beiden V2 Boxen für 10 Sekunden vom Strom getrennt habe hat der Mini SE einwandfrei geladen.

@peterpoetzi @0xFEEDC0DE64

Wann kann ich mich darauf verlassen, dass ich mit den go-e Produkten einwandfrei laden kann?

fuchs-1978, 2021-09-25:
Ich habe dreimal die gleiche Version mit gleicher Software. Mit einem Mii und einem EQV noch nie Probleme gehabt.

kaeferfreund, 2021-09-25:
Die gleiche Version heißt das Modell vom Typ go-e V2?

fuchs-1978, 2021-09-25:
Genau, die V2 ohne die Phasenumschaltung. Ebenfalls mit der 41er Version. Läuft perfekt im PV Überschuss, es wird also mehrmals über alw das Laden ein und ausgeschaltet. Über AMP setze ich die Ströme

kaeferfreund, 2021-09-25:
Ja! Jedes Auto mit dem Kabel das ab Werk geliefert wurde. Also 3 verschiedene Fahrzeuge mit 3 verschiedenen Kabeln probiert.

Nochmal: Das Problem ist nicht mit einem Fahrzeug (und Kabel) Wechsel verschwunden. Nicht mit einem Neustart durch die App sonder NUR nach dem stromlos Schalten.

@fuchs-1978 Ich würde mich nicht wundern, wenn das Problem bei Dir auch noch auftritt, so war es in der vorherigen Firmware auch. Da haben anfangs noch Leute groß erzählt "bei mir tritt das Problem nicht auf". Am Ende waren alle Wallboxen betroffen meinem Bekanntenkreis und das sind dank KFW Förderung nicht wenige. Viele haben es lange nicht gemerkt, weil Sie kein E-Auto hatten oder dachten, dass das periodische Leuchten (Neustart...) gewollt sei...

fuchs-1978, 2021-09-25:
Aber ich habe ja die gleiche Hardware wie du. V2 mit der 41 Version. Und habe zwei Autos an den drei Boxen. Läuft problemlos

kaeferfreund, 2021-09-25:
Ich wünsche jeden, dass das so bleibt. Aber meine Erfahrung hier sagt ganz klar: abwarten, so lange gibts die 041.0 ja noch nicht und bei mir lief Sie ja schon vor der offiziellen Verteilung, weil meine V2 Boxen mit der Vorgänger FW unglaublich viele Abstürze hatten... (die V3 Boxen haben ebenfalls noch viele, viele Bugs...)

chilobo, 2021-09-26:
Ich hatte 2 Monate lang die V2 und dann die V3. Keine Abstürze mit Ioniq und e-up (2018). Beim e-up muss man immer zwei Mal anstecken und er wacht nicht mehr auf, wenn man die Box länger als 20 s ausgeschaltet hat bzw. sie beim PV-gesteuerten Laden abschaltet oder den Strom auf < 6 A stellt. Beim Ioniq keine Probleme. 3-phasiges Laden habe ich noch nicht probiert. Aber das nutzt Dir ja nichts, wenn ich keine Probleme habe ;)

kaeferfreund, 2021-09-28:
Eine merkwürdige Sache ist mir noch aufgefallen: 

Alles hat damit angefangen, dass die API als Momentanverbrauchswert 1 W zurück obwohl kein Fahrzeug angeschlossen ist. Als Eigenverbrauch der Box erscheint mir das zu niedrig und nach dem Neustart zeigt die Box wieder 0W an...

Dieses Phänomen ist mir bereits mehrfach aufgefallen, damit scheint sich ein Absturz anzukündigen.

peterpoetzi, 2021-11-26:
Hi, da ich denke dass es sich hier nur um ein Problem mit einer Box handelt, denke ich dass es sinnvoll ist den Support zu kontaktieren @kaeferfreund

kaeferfreund, 2021-12-01:
Vielen Dank für die schnelle Antwort...

Die Boxen sind mittlerweile ersetzt

closed 2021-12-01 by kaeferfreund