goecharger/go-eCharger-API-v1 #91: ModBus - Verbrauch pro RFID Karte?

futureweb, 2021-10-31:
Hallo,
gibt es iw. die Möglichkeit den gespeicherten Verbrauch pro RFID Karte über ModBus auszulesen?
Über die API ist es ja möglich, nur leider nichts über ModBus gefunden.
Vielen Dank, lg
Andreas S.

ovhelemann, 2023-10-17:
Im Release habe ich das jetzt gelesen, jedoch leider in der Dokumentation die Register noch nicht gefunden. 

[image]

ovhelemann, 2023-10-18:
Der Support teilte mir mit, dass die register im nachfolgenden Link, ganz unten zu finden sind

https://github.com/goecharger/go-eCharger-API-v2/blob/main/modbus-de.md

Vg ovh

0xFEEDC0DE64, 2023-10-25:
genau, los gehts ab register 30333 (332)

dürfen wir dieses Ticket als gefixt betrachten?

closed 2024-06-26 by erxbout