goecharger/go-eCharger-API-v1 #94: Please include information about IP address and MAC address in the app

MathieuWuyts, 2021-11-30:
Hello,
I'm trying to connect the charger in an industrial environment with a lot of switches and devices on the same network, so it is not easy to find the IP address. Please include a function in the app to show the IP address and MAC address of the device when connected to a network.

peterpoetzi, 2021-11-30:
When you are installing a new charger, you probably have the V3 version of the charger. The app actually shows the IP Address already, but not the MAC Address. We will improve this with the next App version. Thank you very much for your suggestion.

For V3 charger you can use the **ccw** (currently connected wifi) api key, which shows:
```
{
    "ssid": "goe-wlan",
    "encryptionType": 3,
    "pairwiseCipher": 4,
    "groupCipher": 4,
    "b": true,
    "g": true,
    "n": true,
    "lr": false,
    "wps": false,
    "ftmResponder": false,
    "ftmInitiator": false,
    "channel": 11,
    "bssid": "BA:FB:11:11:11:11",
    "ip": "192.168.11.11",
    "netmask": "255.255.0.0",
    "gw": "192.168.1.1",
    "ipv6": [
        "FE80::7E9E:BDFF:FE11:1111"
    ],
    "dns0": "192.168.1.1",
    "dns1": "192.168.1.2",
    "dns2": "0.0.0.0"
}
```
as well as to the api keys:
**macs** (mac address station) and
**maca** (mac address access point)

MathieuWuyts, 2021-11-30:
I’m using an older one with V2 hardware. I don’t get any additional info aside from this: 
[image]