goecharger/go-eCharger-API-v1 #95: "Austecken Simulieren" bei aktivem Lastmanagement funktioniert nicht

MichaelFrech, 2021-12-08:
Guten Tag,

kurze Zusammenfassung zur Ausgangssituation: 
2x Go-e Charger V3 (22kw) mit FW 052
Kfz 1: Citroen C Zero BJ 2013
Kfz 2: noch nicht vorhanden 

Beobachtung: 
Da ein weiteres E-Kfz demnächst ankommt, habe ich einen 2ten go-e angeschafft.
Das bestehende Kfz (C-Zero) konnte ich bereits mit der V3 Version des go-e per Ladetimer steuern bzw. manuell Ladungen Stoppen und Starten ( seit V3 mit der Funktion "Ausstecken Simulieren" ist das möglich, mit V2 nicht)
Bevor das 2te Kfz eintrifft, wollte ich bereits die Lastmanagement Funktion aktivieren, um dies dann demnächst zu Testen.
Leider funktioniert mit aktivem Lastmanagement das "Ausstecken Simulieren" nicht mehr zuverlässig bis gar nicht.
Somit kann die Ladetimer Funktion ebenso nicht zuverlässig verwendet werden. 
Nach langwierigen Testversuchen habe ich schlussendlich die Lastmanagement Funktion deaktiviert und danach Funktionierten die genannten Funktionen wieder wie gehabt.

Da ich zu diesem Thema noch sehr wenig gefunden habe, wollte ich Ihnen diese Beobachtung mitteilen um eventuell eine mögliche Fehlerbehebung an zu stoßen.

Beste Grüße 
Michael

christoph-dunker, 2022-01-15:
Das gleiche Problem habe ich auch.

Ich habe über Monate versucht den Grund heraus zu bekommen, warum das "Ausstecken simulieren" nicht funktioniert.
Beim harten ausschalten mittels Sicherung oder beim neustart via app fing das Auto wieder an zu laden.
Als ich diesen Beitrag gelesen habe, habe ich das Lastmanagement abgeschaltet und voila: es funktioniert!

MichaelFrech, 2022-01-15:
Somit haben wir hier schonmal einen kleinen Erfolg für dich (und eventuell auch andere) . Schauen wir mal ob beide Funktionen gleichzeitig seitens go-e Entwickler Team hinzubekommen sind.
Beste Grüße

christoph-dunker, 2022-01-15:
Ich hoffe dass die Entwickler hier auch lesen. Im Zweifelsfall Mal bitte einmal Lebenszeichen geben liebe goE Entwickler.

Ich brauche das Feature Lastmanagement aktuell noch nicht sehr dringend, da ich 22kW darf laut VNB und Sicherung, aber sobald die dritte Box genutzt wird muss ich es wieder aktivieren. Definitiv werde ich aber nicht warten bis der Garantiezeitraum vorbei ist und die Geräte vorher reklamieren und notfalls zurückgeben. Ich hoffe aber, dass es bald gefixt wird und fertig.

copacetic82, 2022-01-21:
Nach langer Fehlersuche habe ich heute vom go-e Support erfahren, dass ein Ladevorgang bei aktivem Lastmanagement nicht beendet und wieder aufgenommen werden kann. Betrifft wohl Unterbrechungen jeglicher Art, auch Phasenumschaltungen. @MichaelFrech Ich denke dein Problem wird damit zu tun haben. 
Bei mir hat es die Auswirkung, dass mein Auto mit voller Batterie während einer Vorklimatisierung den Strom nicht aus dem Charger bezieht, sondern aus der Batterie (Ladevorgang müsste dazu ja erneut gestartet werden). Ich hoffe, dass dieses Verhalten bald korrigiert wird. Ich habe 3 Charger und bin auf das Lastmanagement angewiesen.

christoph-dunker, 2022-01-21:
Geht mir ganz genauso. Wenn das von go e nicht als Fehler behandelt und behoben wird dann muss ich eben vom Kaufvertrag zurücktreten und die Geräte zurück geben. Ich kann ja auch nicht sagen dass das Auto Allrad hat aber dies nur beim Bremsen funktioniert... Nicht alles was hinkt ist auch ein Vergleich, das weiß ich, aber ich fühle mich hier wirklich getäuscht da ich die Wallboxen nur aus diesem Grund gekauft habe. Phasenumschaltung und Lastmanagement benötige ich beides. Und dass das Auto den Strom zum klimatisieren aus der Wallbox auch bei Lastenmanagement nehmen kann sollte klar sein.

Gibt es einen sachlichen Grund außer Software kann nicht beides, warum man Lastmanagement und wieder aufnehmen des Ladens nicht machen kann? Wenn ich das Lastmanagement von außen per Hand steuere dann klappt es doch auch.

peterpoetzi, 2022-01-21:
Ein Fix für diese derzeitige Limitierung des Lastmanagements ist derzeit in Arbeit in wird in den nächsten Wochen getestet und released

christoph-dunker, 2022-01-21:
Sauber, das wollten wir glaube ich alle hören. Daumen hoch und schönes Wochenende!

sigma415, 2022-01-27:
> Ein Fix für diese derzeitige Limitierung des Lastmanagements ist derzeit in Arbeit in wird in den nächsten Wochen getestet und released

Das ist schön, dass ein Fix in Entwicklung ist. Auch ich habe wochenlang am Last-Mgt rumgedoktert, da nach Vorklimatisierung ohne Stecker raus/rein oder Box-Reset nicht mehr geladen wurde.
**Wird es dann (hoffentlich) auch für die alte Hardware V2 eine neue Firmware geben ?**

copacetic82, 2022-03-19:
> Ein Fix für diese derzeitige Limitierung des Lastmanagements ist derzeit in Arbeit in wird in den nächsten Wochen getestet und released

Ist das nach wie vor geplant, oder gilt das nur für neue HW? Ich habe 3x Charger V3 mit 0.53.1.

Falls das für ältere HW nicht mehr vorgesehen ist, bitte ich um Rückinfo. Dann muss ich mir was eigenes programmieren. Würde ich mir aber gerne ersparen.

sigma415, 2022-03-19:
@peterpoetzi: Wann dürfen wir mit dem Fix für V2 und V3 rechnen ?

Headroom42, 2022-05-09:
@peterpoetzi Gibt es schon ein Update wann der Fix eingebaut ist? Habe einige Tage gesucht bis ich auf diesen Thread gestossen bin, da das lastgesteuerte PV Laden nicht funktioniert hat bei gleichzeitig eingeschalteten Load Balancing. Ich würde gerne Beides nutzen können.

fuchs-1978, 2022-09-29:
> 

Für die V2 macht das doch keinen Sinn, die V2 kann nicht zwischen einer und drei Phasen wechseln. Daher wird kein ausstecken benötigt

sigma415, 2022-09-29:
Mein Problem hat nichts mit 1ph/3ph-Umschaltung zu tun sondern: siehe mein comment oben vom 27 Jan
[https://github.com/goecharger/go-eCharger-API-v1/issues/95#issuecomment-1023341122](url)

fuchs-1978, 2022-09-29:
> Umschaltung

Alles, ich dachte auf den Eingangspost, sorry

sigma415, 2023-06-22:
> Ein Fix für diese derzeitige Limitierung des Lastmanagements ist derzeit in Arbeit in wird in den nächsten Wochen getestet und released

Ist dieser von Peter Pötzi vor langem versprochene Lademgt-Fix für die HW V2 in der neuen Firmware 0.42 enthalten ?

Mein App zeigt seit neuestem ein FW-Update auf 0.42 an, leider finde ich nirgends eine Info über die FW 0.42, geschweige denn ein Change Log.
Kennt jemand eine Info-Quelle zur FW 0.42 ?