goecharger/go-eCharger-API-v1 #96: Frage zu dem Stromstufen (6 Ampere aufwärts)

klacol, 2021-12-14:
Hallo, 

ich setze den go-eCharger zuverlässig ein, auch in Kombination mit Home-Assistant, dank der Integration [homeassistant-goecharger](https://github.com/cathiele/homeassistant-goecharger) .

Ich habe mir in Home-Assistant einen Überlastungsschutz gebaut, indem ich den Charger automatisch auf einen reduzierten Ladestrom stelle, wenn meinem Hausanschluss die Überlastung droht (Durchlauferhitzer). Nun arbeite ich an einem HA-basierten PV-Überschussladen. Ich habe eines aber noch nicht verstanden und frage einfach mal hier nach.

Aus welchem Grund hat der Charger die verschiedenen Stufen in der Stromstärke? Es fängt bei 6 Ampere an und geht dann in Stufen hoch. 

* Warum kann man das nicht stufenlos machen?
* Warum gibt es keine kleinere Stufe als 6 Ampere?

Das hat sicher eine Begründung, evtl. irgendwo in der Elektrotechnik. Kann das jemand erklären?

Danke vorab.

fuchs-1978, 2021-12-14:
Kleiner als 6 A ist laut der Norm nicht vorgesehen. Du kannst aber ein einphasiges Kabel nehmen, so hast du nur noch ein Drittel der Leistung

fuchs-1978, 2021-12-14:
Stufenlos wurde hier Mal besprochen, das ist wohl Hardware seitig nicht möglich

klacol, 2021-12-15:
ok, danke

closed 2021-12-15 by klacol