goecharger/go-eCharger-API-v1 #104: Modbus 1/3 Phasenumschaltung - Hardware V3

JoGi65, 2022-02-11:
Wird es jemals ein Modbus Register für die Phasenumschaltung 1/3 geben oder nicht - Hardware V3?

Bitte eine klare Aussage!

Es ist für mich eine wichtige Funktion, und ich brauche für unser entstehendes PV Carport eine zweite BOX.

christianmw24, 2022-03-29:
Wöre auch für mich eine hilfreiche Info bzw. erwünschte Funktion. Soweit ich die Dokumentation verfolgt habe, gibt es akutell weder über Modbus noch über die http-API bzw. MQTT die Funktion der Phasenumschaltung. Für wirklich sinnvolle PV-Nutzung wäre eine Phasenumschaltung wirklich hilfreich. Manuell in der App umstellen ist doch etwas umständlich.
Würde mich auch über eine Aussage des Herstellers freuen, ob eine Phasenumschaltung via Modbus geplant ist.

mincequi, 2022-10-26:
Gibt es hierzu ein Update? Phasenumschaltung per Modus wäre sehr wünschenswert.

fuchs-1978, 2022-10-26:
http://10.20.60.21/api/set?psm=2

http://10.20.60.21/api/set?psm=1

Über API geht es

JoGi65, 2022-10-26:
Deswegen ist es umso verwunderlicher, dass das Register nicht bekannt gegeben wird, bzw. keine Antwort erfolgt.

fu-zhou, 2022-11-21:
bräuchte das auch, ein Mix aus MQTT-, HTTP- und Modbus-Befehlen ist lästig und trägt nicht zur Transparenz bei...

JoGi65, 2023-06-08:
40333 (332)	PHASE_SWITCH_MODE	Holding Register	uint16_t	1	Selbe wie api key psm, akzeptiert nur 0, 1, 2 (Seit Firmware 55.5)

Schreiben funktioniert, Lesen kommt nix, soweit ich sehe (55.6 Beta)

ovhelemann, 2023-08-17:
Danke @JoGi65 für das Register. Hatte ich bisher nicht eingebunden. Bei mir funktioniert lesen und schreiben.