goecharger/go-eCharger-API-v1 #106: Phase switching via API

pmihairo, 2022-07-03:
Hello, 

A very important function for me is phase switching. Use case: 
- Home with grid connected PV system
- Go-e charger maximum current per phase controlled remotely based on the amount of power generated by the PV system and fed to the grid.
- Once power from PV system fed to grid drops below 6A per phase, charging has to stop (do not want to use power from grid). 

- Would like to be able to set 1 phase mode so that charging can continue even if the system generates less than 6A per phase total. 

Thanks,

archie-bal, 2022-09-08:
Yes, I would like to switch phase modes also, with API v1. The parameter "pha" seems to be read-only. Is this true?

marcelgsteiger, 2023-02-08:
FYI, I'm using API v2 with the psm parameter (r/w) to switch between 3 and 1 phase using node red. As soon as the available solar power exceeds some percentage of the demand for all currently charging stations, I switch from 1 to 3-phase, with some hysteresis to switch back. Moreover, I always force 3-phase charging when the cheap night energy tariff starts (here at 9 pm). The go-e are connected to the grid with rotating phases to minimize asymmetric phase load when multiple cars are charging simultaneously in one-phase mode.