goecharger/go-eCharger-API-v1 #114: GO-E V2 mit Firmware 042.0: amx Parameter nicht mehr schreibbar?

mdahinden, 2023-06-23:
Seit dem Update auf die neue Firmware kann ich den `amx`-Parameter nicht mehr via MQTT setzen (`amp` geht). Wurde dieser entfernt/schreibgeschützt? 

Falls ja: welchen Parameter kann ich stattdessen nutzen, um via HomeAssistant den aktuellen PV-Strompegel zu setzen? Am `amx`-Parameter hat mir gefallen, dass dieser nicht auf dem Flash persistiert wurde (Zitat Dokumentation: "Wird nicht auf dem Flash persistiert, verhält sich sonst aber gleich wie amp. [...] Nutzung von amx für PV Regelung empfohlen. ")

PhilippFoerster, 2023-06-24:
Ich hab dasselbe Problem seit der neusten Firmware. Ich hab ein bisschen getestet und einen Workaround gefunden: Es scheint so, als könnte man amx keinen Wert > amp zuweisen. Heißt: Wenn du amp auf 16 setzt, dann kann amx auch Werte von 6 bis 16 annehmen. Setzt du dann amx = 8 werden z.b. trotzdem nur ~5,4kW gezogen. Allerdings steht in der App dann immer, dass mit 16 Ampere geladen wird...

guruguruguru, 2023-06-26:
Ich habe dsa selbe Problem mit der neuen Firmware Version. Es scheint zwar der korrekt Wert via amx gesetzt zu werden aber beim auslesen aus dem amp Wert stimmt dieser dann nicht.

Den amp Wert will ich aus dem oben genannten Grund (persistentes Schreiben) eigentlich nicht überschreiben.

guruguruguru, 2023-06-26:
FYI: Ich habe dem Support geschrieben und das Angebot bekommen vorerst auf die letzte FW-Version zurückzugehen. Das werde ich auch tun, damit die PV Steuerung wieder anstandslos funktioniert.

mdahinden, 2023-06-26:
@guruguruguru Danke für die Information. Wurde dir auch gesagt, wie man eine alte FW-Version installieren kann?

guruguruguru, 2023-06-27:
@mdahinden Der Support hat mir das angeboten, ich musste die Resetcard abfotografieren und schicken. Bisher ist aber noch nichts passiert.

chk2902, 2023-07-01:
Betrifft übrigens - nicht erstaunlich - nicht nur die Steuerung per MQTT, sondern auch HTTP.

marvkis, 2023-07-02:
Hi. Mich hat das Thema auch 'erwischt' Der Hint @PhilippFoerster mit dem amp als 'max' ist schon mal Gold wert, der stand bei mir nur auf 6A. Jetzt hab ich gerade gesehen das im MQTT-Json der amx-Wert zurück geliefert wird, und auch passend aussieht. ich hab leider kein altes json mehr da aber kann es evtl. sein das es vor der 0.42 keinen amx wert gab und nur den amp? @guruguruguru kannst Du das nach erfolgten Downgrade evtl. mal verifizieren?

Ich hab bei mir gerade ein paar Tests gemacht, und zumindest hat die Station jetzt auf manuelle amx-Änderungen reagiert, allerdings amp und der wert in der App zeigen jetzt immer nur die letzte 'amp'-Einstellung von der APP an. Allerdings hat die Ladeleistung variiert. Und imho war es in der Vergangenheit auch so das die "eingestellte Leistung" in der App den amx gespiegelt hatte, bin mir aber nicht 100% sicher.. 🤔

mdahinden, 2023-07-02:
Ich habe jetzt ein Paar Tage die von @PhilippFoerster erwähnte Einstellung im Einsatz (amp auf 16A, gewünschte Ladeleistung per amx gesetzt). Damit funktioniert die PV Steuerung bei mir wieder einwandfrei. Unschön ist lediglich, dass in der App auch bei mir immer die maximale Ladeleistung angezeigt wird. Das war vor dem Update auf 0.42 nicht so. Ich konnte jeweils in der App kontrollieren, ob die aktuelle Ladeleistung per amx korrekt gesetzt ist.

@marvkis In der mqtt-Message sehe ich mit 0.42 beide Werte: amp und amx. Das war auch schon vor dem Update so. Ersterer steht auf 16A, Letzterer zeigt die per MQTT gesetzte Ladeleistung.  Wenn ich mich richtig erinnere, wurde vor dem Update jeweils auch die per amx gesetzte Ladeleistung bei amp angezeigt (ich hoffe doch, dass diese dort bloss angezeigt, aber nicht tatsächlich persistiert wurde).

@guruguruguru Danke für die Hinweis. Das mache ich!

chk2902, 2023-07-03:
@FEEDC0DE64 kann man das Problem "dringend", sprich zur "breaking change" machen? Ich bekomme immer mehr Kunden, die den go-eCharger V2 nicht nutzen können deswegen.

myModelY, 2023-07-08:
warum soll amp den flash killen? was macht die app von go-e anders? die schreibt acu h nur amp um

fuchs-1978, 2023-07-08:
AMP wird immer in den flash geschrieben, AMX nicht. Wenn man also jede 5 Sekunden schreibt ist ohne flash besser. Mit der App verstellst du ja nicht alle 5 Sekunden

myModelY, 2023-07-08:
aha. wie macht das dann go-e in verbindung mit awattar?
und wer schreibt alle 5sec. um?

fuchs-1978, 2023-07-08:
Kann sein, dass sie mit awattar AMP nutzen, das muss aber auch nicht so oft geschrieben werden. Der Preis ändert sich ja nur einmal die Stunde. Ich schreibe den Wert alle 5 Sekunden um die Autos dynamisch an die PV Leistung anzupassen

myModelY, 2023-07-08:
und warum dann diese app nicht?  nur weil du alke 5sek. schreiben willst?

fuchs-1978, 2023-07-08:
Weil der flash eine begrenzte Lebensdauer hat, abhängig von den Schreibzugriffen. Da ist es halt ein Unterschied ob du einmal in der Stunde schreibst oder alle 5 Sekunden. Mit PV musst du halt häufig nachregeln

chk2902, 2023-07-27:
Das Problem der 042.0 ist weiterhin vorhanden. Hallo, go-e? Jemand zuhause?

wrzlbrmft1, 2023-08-11:
Hi!
ich hab das gleiche Problem, dass das Flag "amx" nicht gesetzt wird. Hab ein wenig mit "amp" rumgespielt und festgestellt, dass der "amp" den "amx" begrenzt:
-> amx=16 und amp=6 --> es werden 6 A eingestellt
-> amx=16 und amp=16 -> es werden 16 A eingestellt

nicht schön - aber ich glaube, so lässt sich das Problem umgehen . . . .

chk2902, 2023-08-11:
Klar. Zwar zeigt dann die go-e App an, dass Du mit 6 A (amx) lädtst, aber 16 eingestellt hast (amp), aber das ist deren Problem.

PhilippFoerster, 2023-08-23:
@IPSCoyote Nein, du stellst einmal amp auf 16 und kannst dann weiterhin mit amx regeln. amp dient basically als max Wert

IPSCoyote, 2023-08-23:
@PhilippFoerster 
Verstehe ich das richtig: Man updated auf die 042.0, amx geht nicht mehr, dann toggelt man mit amp=6 und amp=16 den max wert einfach mal, und danach geht amx=xyz wieder?
Was ist denn das für ein ..... Es gilt mal wieder: "never change a running system" (=never update a go-eCharger without the need to)

PhilippFoerster, 2023-08-23:
@IPSCoyote Nicht ganz. amx nimmt einfach keine Werte > amp an. Also ist der "praktischste" move, amp einmalig auf 16 zu setzten, damit du per amx Werte von 6 - 16 setzen kannst. Die werden dann auch beachtet. 
Allerdings siehst du halt in der App immer, dass mit 16A geladen wird (falsch) und du solltest das Steuern per App unterlassen, da hier amp gesetzt wird.

IPSCoyote, 2023-08-23:
@PhilippFoerster Danke für die Erklärung. 

Bei meinen Users (ich habe eine Smarthome-Integration geschrieben) tritt das Problem nach dem FW Update auf. Die haben aber vorher alle mit amx geregelt und amp auf 16A stehen gehabt. Und plötzlich geht amx nicht mehr. Da scheint beim Update dan nirgend etwas intern schief zu laufen, so daß amx trotz höherem amp erst mal nicht mehr geht (Vermutung).

Die Firmware sollte aus meiner Sicht dringend gefixt werden. Das ist doch kein Zustand.

IPSCoyote, 2023-08-23:
@PhilippFoerster So ein Mist. Ich empfehle den Usern auch den Downgrade anzufordern. Vielleicht wachen die bei Go-e ja auf, wenn immer mehr danach fragen....

sigma415, 2024-04-11:
Leider tut go-e hier seit ca. 10 Monaten gar nichts und sie verteilen immer noch die fehlerhaft V0.42 und es tappen immer mehr HW V2-Kunden in diese gemeine und unsichtbare Falle.
Siehe: https://github.com/evcc-io/evcc/discussions/8582

myModelY, 2024-04-12:
falsch. die steuerung klappt, wird aber nur falsch angezeigt. man muss den wert ama einstellen

chk2902, 2024-04-12:
Es gibt ja den Workaround, bei der 042.0 den "amp"-Wert immer mindestens auf dem "amx"-Wert zu setzen. Damit klappt's bei meinen Kunden.

Nur ist wie oben erwähnt die Anzeige in der go-e App falsch, aber damit habe ich kein Problem.

mdahinden, 2024-04-12:
@myModelY gemäss Manual empfiehlt es sich nicht `amp` für die dynamische Ladeleistungsregelung zu verwenden, da dieser Wert auf dem Flash-Speicher persistiert wird. Und da dieser Speicher nur eine begrenzte Anzahl an Schreibvorgängen erlaubt, würde dadurch die Ladestation möglicherweise vorzeitig kaputt gehen. Somit besser wie von @chk2902 erwähnt `amx` dynamisch setzen und `amp` permanent auf dem Maximum lassen. Leider sieht man dadurch in der App nur den `amp`-Maximalwert und nicht den tatsächlichen Wert. Deshalb würde ich eine V0.43 für die HW V2 auch sehr begrüssen!

Wolfgang1966, 2024-04-12:
Ich verwende problemlos den Go-E-Charger für Überschussladen wie oben beschrieben. Der Charger macht genau was er soll, auf dem REST Interface passen die Werte auch. Von daher halte ich den "Fehler" für verschmerzbar. Ok, ein Fix sollte jetzt eigentlich auch nicht so kompliziert sein, dass er Monate zur Implementierung braucht :-)